srsran/srsRAN_4G #73: Raspberry Pi 3B

victorstewart, 2017-03-25:
Performance wise, is it possible to run an srsLTE eNodeB on the new raspberry pi 3b (assuming either bladeRF or B210)? Also, it only has USB 2.0.

here are it's quick stats:
https://www.raspberrypi.org/products/raspberry-pi-3-model-b/

victorstewart, 2017-03-26:
ended up choosing Intel Joule. It's 60x faster on sysbench benchmarks, to say the least. Might still be a good discussion though for others interested... could mention what other gear I'll be using?

enesunal, 2017-03-28:
I think depending on the bandwidth, weaker CPUs may or may not be used.

Since PHY layer is running on CPU, there is a high processing power usage with srsLTE while doing signal processing (fft, demodulation, etc.) Also, CPUs with AVX and/or SSE4.1 are needed to do this signal processing fast. I do not think that ARM (Raspberry Pi) or Atom would be able to handle it.

The first solution came to my mind is to use NB-IoT standard if it's supported by your service provider. I'm not sure if NB-IoT is implemented in srsLTE, or if there is a plan to do it.

victorstewart, 2017-03-28:
Those are some solid points. I guess I'll just have to try and see what happens then, fingers crossed.

Trying to operate at 1900MHZ with AT&T. Using a BladeRF.

Are there any modules that come to mind I could mod the intel joule with to give it some more signal processing horsepower?

It seems to have been designed with the intention of expansion... Re GPIO pins etc

Or maybe the there's a different board/chip that would work better? Trying to build it self-contained / portable... so I don't have to use my laptop.

victorstewart, 2017-03-28:
I put in some questions with Intel about the signal processing capabilities of the Joule.

Also found these boards that are touted to be designed for signal processing? This is all new territory for me, but I'm poking around.

https://www.xilinx.com/products/boards-and-kits/board-functions/nav-digital-signal-processing.html

victorstewart, 2017-03-28:
worst case, can't any necessary DSP just be handled on the FPGA?

enesunal, 2017-03-29:
Doing the signal processing outside of the CPU is challenging. Microchips other than you have found, there is SoC (System-on-chip) variations. Also, EttusResearch has a ongoing project about doing "heavy" works on FPGA. They have some ready-to-use FPGA modules and stable connection with the ARM processor. I once tried it to show spectrum with E310 model USRP. The project called RFNoC, but again, you should port some parts of srsLTE to FPGA (by using ready-to-use modules or by getting your hands dirty)

For more information, please check [RFNoC project](https://www.ettus.com/sdr-software/detail/rf-network-on-chip). This is not where you should look for your answers.

victorstewart, 2017-03-29:
awesome, thanks for all the help! I really appreciate it. and sorry for the bit of off topic-ness lol.

P.S. ended up discovering intel NUCs (thank god) went with this: http://www.intel.com/content/www/us/en/nuc/nuc-kit-nuc7i7bnh.html

closed 2017-03-29 by victorstewart

victorstewart, 2017-03-29:
enesunal, is there anyway we could exchange some emails, and I'll compensate you for your time? 

I'd love some helping making sense of srsLTE. I'm very knowledgable in terms of computers, just know little about radio systems or LTE (beyond high level generalities). I just need to get my bearings.. like what the puzzle pieces are, and how they all fit together.