srsran/srsRAN_4G #74: Support for UHD's new logging API

enesunal, 2017-03-27:
A month ago, EttusResearch brought [new logging API replacing msg API](https://github.com/EttusResearch/uhd/commit/21aad77c9ca07f4271136b9241f5adb00a6bb908), since then builds with master branch of the UHD lib is failing with:

```
~/dev/srsLTE/srslte/lib/rf/uhd_c_api.cpp:14:29: error: variable or field ‘translate_handler’ declared void
 void translate_handler(uhd::msg::type_t type, const std::string & msg)
                             ^
~/dev/srsLTE/srslte/lib/rf/uhd_c_api.cpp:14:29: error: ‘uhd::msg’ has not been declared
~/dev/srsLTE/srslte/lib/rf/uhd_c_api.cpp:14:47: error: expected primary-expression before ‘const’
 void translate_handler(uhd::msg::type_t type, const std::string & msg)
                                               ^
~/dev/srsLTE/srslte/lib/rf/uhd_c_api.cpp: In function ‘void rf_uhd_register_msg_handler_c(void (*)(const char*))’:
~/dev/srsLTE/srslte/lib/rf/uhd_c_api.cpp:23:8: error: ‘uhd::msg’ has not been declared
   uhd::msg::register_handler(translate_handler);
        ^
~/dev/srsLTE/srslte/lib/rf/uhd_c_api.cpp:23:30: error: ‘translate_handler’ was not declared in this scope
   uhd::msg::register_handler(translate_handler);
                              ^
srslte/lib/rf/CMakeFiles/srslte_rf.dir/build.make:134: recipe for target 'srslte/lib/rf/CMakeFiles/srslte_rf.dir/uhd_c_api.cpp.o' failed
make[2]: *** [srslte/lib/rf/CMakeFiles/srslte_rf.dir/uhd_c_api.cpp.o] Error 1
CMakeFiles/Makefile2:2662: recipe for target 'srslte/lib/rf/CMakeFiles/srslte_rf.dir/all' failed
make[1]: *** [srslte/lib/rf/CMakeFiles/srslte_rf.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

I'm leaving a note for the googlers, you can use UHD 3.9 LTS branch for srsLTE until the new version of UHD is released.

ismagom, 2017-03-27:
I just pushed a fix for this today. Can you check it again?

enesunal, 2017-03-28:
Fix(commit: [cbb7e3c](https://github.com/srsLTE/srsLTE/commit/cbb7e3cea0b925f60c7eefe1b8fce58b2b89f5ff)) is working well, thanks.

closed 2017-03-28 by enesunal