srsran/srsRAN_4G #77: Compilation on non-Intel Ubuntu machines?

HughAndersonNUS, 2017-04-07:
Hi - I am not sure if this is an issue, or a request, but I wanted to run srsLTE on a small fast ARM board (The Odroid XU4), and using limeSDR. The processor board runs other gnuradio based software successfully, and soapy seems to work with UHD software (like tx_samples_from_file).
[messages.txt](https://github.com/srsLTE/srsLTE/files/904658/messages.txt)

The compile proceeds with no issues, but when I run the first test example, I get a Bus Error: 
I guess my question is: Should I give up now, or is there some possibility that, through some debugging, this set of hardware could run srsLTE?
Cheers Hugh
I have attached a file showing the output of the first example command:
`odroid@odroid:~/srsLTE-master/build$ ./srslte/examples/pdsch_ue -f 788000000 -d
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Opening RF device...
Opening USRP with args: 
-- Make connection: 'LimeSDR-USB [USB 2.0] 9060B00492A2A'
Estimated reference clock 30.7196 MHz
Selected reference clock 30.720 MHz
-- Device name: LimeSDR-USB
-- Reference: 30.72 MHz
-- Init LMS7002M(0)
LMS7002M values cache at /home/odroid/.limesuite/LMS7002M_cache_values.db
-- Ver=7, Rev=1, Mask=1
-- LMS7002M calibration values caching Enable
CGEN: Freq=80 MHz, VCO=2.56 GHz, INT=82, FRAC=349525, DIV_OUTCH_CGEN=15
-- SoapyLMS7::setFrequency(Rx, 0, BB, 0 MHz)
-- SoapyLMS7::setFrequency(Tx, 0, BB, 0 MHz)
-- SoapyLMS7::setAntenna(Rx, 0, LNAL)
-- SoapyLMS7::setAntenna(Tx, 0, BAND1)
-- SoapyLMS7::setGain(Rx, 0, PGA, 0 dB)
-- SoapyLMS7::setGain(Rx, 0, LNA, 0 dB)
-- SoapyLMS7::setGain(Rx, 0, TIA, 0 dB)
-- SoapyLMS7::setGain(Tx, 0, PAD, -50 dB)
-- SoapyLMS7::setSampleRate(Rx, 0, 10 MHz), CG, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
----- FPGA PLL #1 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 114.2 (steps 132)
----- FPGA PLL #0 CONFIG -----
M=182, N=7, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 100.4 (steps 58)
-- SoapyLMS7::setSampleRate(Tx, 0, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
----- FPGA PLL #1 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 114.2 (steps 132)
----- FPGA PLL #0 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 95.2 (steps 110)
-- SoapyLMS7::setBandwidth(Rx, 0, 30 MHz)
-- Rx Filter calibrated from cache
-- SoapyLMS7::setBandwidth(Tx, 0, 30 MHz)
-- Tx Filter calibrated from cache
-- SoapyLMS7::setFrequency(Rx, 1, BB, 0 MHz)
-- SoapyLMS7::setFrequency(Tx, 1, BB, 0 MHz)
-- SoapyLMS7::setAntenna(Rx, 1, LNAL)
-- SoapyLMS7::setAntenna(Tx, 1, BAND1)
-- SoapyLMS7::setGain(Rx, 1, PGA, 0 dB)
-- SoapyLMS7::setGain(Rx, 1, LNA, 0 dB)
-- SoapyLMS7::setGain(Rx, 1, TIA, 0 dB)
-- SoapyLMS7::setGain(Tx, 1, PAD, -50 dB)
-- SoapyLMS7::setSampleRate(Rx, 1, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
----- FPGA PLL #1 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 114.2 (steps 132)
----- FPGA PLL #0 CONFIG -----
M=182, N=7, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 100.4 (steps 58)
-- SoapyLMS7::setSampleRate(Tx, 1, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
----- FPGA PLL #1 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 114.2 (steps 132)
----- FPGA PLL #0 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 95.2 (steps 110)
-- SoapyLMS7::setBandwidth(Rx, 1, 30 MHz)
-- Rx Filter calibrated from cache
-- SoapyLMS7::setBandwidth(Tx, 1, 30 MHz)
-- Tx Filter calibrated from cache
Starting AGC thread...
-- SoapyLMS7::setGain(Rx, 0, LNA, 30 dB)
-- SoapyLMS7::setGain(Rx, 0, TIA, 12 dB)
-- SoapyLMS7::setGain(Rx, 0, PGA, 8 dB)
CGEN: Freq=30.72 MHz, VCO=2.4576 GHz, INT=79, FRAC=0, DIV_OUTCH_CGEN=39
Tunning receiver to 788.000 MHz
-- SoapyLMS7::setFrequency(Rx, 0, RF, 788 MHz)
SetFrequency using cache values vco:1, csw:196

UHD Warning:
    GetDC_IQ_Interp(788 MHz, ch=0, tx=0): no matches between [787, 789] MHz
-- SoapyLMS7::setFrequency(Rx, 0, BB, -4.88281e-06 MHz)
Searching for cell...
Warning popping from TX, samples popped 0/1360
Using Volk machine: neon_hardfp_orc
Warning popping from TX, samples popped 0/1360
-- SoapyLMS7::setSampleRate(Rx, 0, 1.92 MHz), CGEN=30.72 MHz, ADC=7.68 MHz, decim=4
ConnectionSTREAM::ConfigureFPGA_PLL(tx=3.84MHz, rx=1.92MHz)
Warning popping from TX, samples popped 0/1360
----- FPGA PLL #0 CONFIG -----
M=136, N=1, Fvco=1044.480 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 91.7 (steps 277)
	Rx pktLoss@4 - ts diff: 1232507136  pktLoss: 0.0
	Rx pktLoss@7 - ts diff: 1206153216  pktLoss: 0.0
Bus error
Hughs-MacBook-Pro:~ hugh$ 
`

suttonpd, 2017-04-10:
Hi Hugh,
Soapy supports applications which already use the UHD drivers but we haven't tested this approach ourselves with any SRS software.
That being said, we are currently working with both the Odroid XU4 and limeSDR and expect to support both soon.

Regards,
Paul

closed 2017-04-10 by suttonpd

HughAndersonNUS, 2017-04-11:
Hi Paul,

Thanks for that. I look forward to this. Any idea of a timescale?

Cheers Hugh

Hugh Anderson                             E-mail: hugh@comp.nus.edu.sg
SoC, National University of Singapore     http://www.comp.nus.edu.sg/~hugh

HughAndersonNUS, 2017-05-11:
Hi Paul,
I have tried out srsLTE-next with my limeSDR and odroid XU4. I had one tiny teething trouble getting cmake to find the LimeSuite library: I had to add /usr/lib/arm-linux-gnueabihf to the srsLTE-next/cmake/modules/FindLimeSDR.cmake file:

```
find_library(LIMESDR_LIBRARIES 
    NAMES LimeSuite
    PATHS ${LIMESDR_PKG_LIBRARY_DIRS}
          /usr/lib/arm-linux-gnueabihf
          /usr/lib
          /usr/local/lib
    }
```
After doing that it compiled cleanly. However, it does not seem to work for me. For example, If I try to run `cell_search`, I end up like 
```
-- SoapyLMS7::setGain(Rx, 0, LNA, 30 dB)
-- SoapyLMS7::setGain(Rx, 0, TIA, 12 dB)
-- SoapyLMS7::setGain(Rx, 0, PGA, 19 dB)
CGEN: Freq=30.72 MHz, VCO=2.4576 GHz, INT=79, FRAC=0, DIV_OUTCH_CGEN=39
SetFrequency using cache values vco:2, csw:157
[INFO]:  Set rf_freq to 1805.000 MHz
[  0/749]: EARFCN 1200 Freq. 1805.00 MHz looking for PSS.

Using Volk machine: neon_hardfp_orc
Warning popping from TX, samples popped 0/1360
[INFO]:  Setting sampling frequency 1.92 MHz for PSS search
ConnectionSTREAM::ConfigureFPGA_PLL(tx=3.84MHz, rx=1.92MHz)
----- FPGA PLL #0 CONFIG -----
M=136, N=1, Fvco=1044.480 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 91.7 (steps 277)
[INFO]:  Starting receiver...
        Rx pktLoss@0 - ts diff: 1227108352  pktLoss: 0.0
        Rx pktLoss@3 - ts diff: 1190420480  pktLoss: 0.0
Warning popping from TX, samples popped 0/1360
Warning popping from TX, samples popped 0/1360
Bus error
```

My system seems to recognize the LimeSDR device OK, and I can use it with other gnuradio tools:
```
odroid@odroid:~/srsLTE-next/build/srslte/examples$ SoapySDRUtil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Found device 0
  addr = 1d50:6108
  driver = lime
  label = LimeSDR-USB [USB 3.0] 9060B00492A2A
  media = USB 3.0
  module = STREAM
  name = LimeSDR-USB
  serial = 00090XXXXXXX

```
Any idea where I can go from here?

yagoda, 2017-05-16:
Hi!

The support for the LimeSDR is more stable using the SoapySDR API.

If you have Soapy installed you can pull the latest version of srsLTE/next and recompile and the cmake should now find the Soapy binaries and use them instead of the native Lime drivers.

Also make sure you have calibrated the your Lime board for the specific frequencies you wish to use by using the "LimeSuiteCalibrate" script that should be located in your /usr/bin folder.

Regards,
Justin

HughAndersonNUS, 2017-05-16:
Hi Justin,
I *think* it is using the Soapy SDR. When I run cell_search, for example, it shows:
 ```
  -- SoapyLMS7::setGain(Rx, 0, LNA, 30 dB)
  -- SoapyLMS7::setGain(Rx, 0, TIA, 12 dB)
```
srsLTE-next works fine for me using an Ubuntu x86 machine, but I wanted to use the Odroid XU4, and it reliably gives me the bus error in my post.
Cheers Hugh

yagoda, 2017-05-16:
Hey Hugh,

I am having trouble reproducing your error, could you please delete your build folder, rerun cmake and post the output. Also, could you try running the following:
_sudo taskset 0xF0 ./pdsch_ue -f  930000000 -g 0.8_

and send me the output of this also.

Be sure to LimeSuiteCalibrate for these frequencies before you run this command.

Thanks!

Justin

HughAndersonNUS, 2018-11-02:
Hi - just got back to this after a year or so, and upgraded everything to the latest version of srsLTE, and tried recompiling on ARM/Ubuntu 18.04.1. Unfortunately, the build fails with the following SSE-related error:
`[ 10%] Building C object lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o
/home/odroid/srsLTE/lib/src/phy/fec/turbodecoder.c: In function ‘srslte_tdec_init_manual’:
/home/odroid/srsLTE/lib/src/phy/fec/turbodecoder.c:170:22: error: ‘sse16_win_impl’ undeclared (first use in this function); did you mean ‘sse_impl’?
       h->dec16[0] = &sse16_win_impl;
                      ^~~~~~~~~~~~~~
                      sse_impl
/home/odroid/srsLTE/lib/src/phy/fec/turbodecoder.c:170:22: note: each undeclared identifier is reported only once for each function it appears in
/home/odroid/srsLTE/lib/src/phy/fec/turbodecoder.c:178:21: error: ‘sse8_win_impl’ undeclared (first use in this function); did you mean ‘sse16_win_impl’?
       h->dec8[0] = &sse8_win_impl;
                     ^~~~~~~~~~~~~
                     sse16_win_impl
lib/src/phy/fec/CMakeFiles/srslte_fec.dir/build.make:302: recipe for target 'lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o' failed
`
Is there some cmake/make option for forcing the SSE software-only build?

Cheers Hugh