srsran/srsRAN_4G #80: Sync Test Failing

sirjansel0t, 2017-06-09:
Hello, I ran make test and I am getting these results:

*** stack smashing detected ***: /home/cewgroup/srsLTE/build/lib/src/phy/sync/test/sync_test terminated
 93/131 Test  #93: sync_test_100 ....................***Exception: Other  0.88 sec
        Start  94: sync_test_400
*** stack smashing detected ***: /home/cewgroup/srsLTE/build/lib/src/phy/sync/test/sync_test terminated
 94/131 Test  #94: sync_test_400 ....................***Exception: Other  0.87 sec
        Start  95: sync_test_100_e
*** stack smashing detected ***: /home/cewgroup/srsLTE/build/lib/src/phy/sync/test/sync_test terminated
 95/131 Test  #95: sync_test_100_e ..................***Exception: Other  0.87 sec
        Start  96: sync_test_400_e
*** stack smashing detected ***: /home/cewgroup/srsLTE/build/lib/src/phy/sync/test/sync_test terminated
 96/131 Test  #96: sync_test_400_e ..................***Exception: Other  0.88 sec

The sync function seems to fail.  This is with SRSLTE 2.0.

sirjansel0t, 2017-06-09:
After looking through make these warnings came up:

/home/cewgroup/srsLTE/lib/src/phy/utils/vector.c: In function ‘srslte_vec_max_fi’:
/home/cewgroup/srsLTE/lib/src/phy/utils/vector.c:722:26: warning: passing argument 1 of ‘volk_32f_index_max_16u’ from incompatible pointer type [-Wincompatible-pointer-types]
   volk_32f_index_max_16u(&target,x,len);
                          ^
/home/cewgroup/srsLTE/lib/src/phy/utils/vector.c:722:26: note: expected ‘unsigned int *’ but argument is of type ‘uint16_t ** {aka short unsigned int **}’
/home/cewgroup/srsLTE/lib/src/phy/utils/vector.c:723:10: warning: return makes integer from pointer without a cast [-Wint-conversion]
   return target;
          ^
/home/cewgroup/srsLTE/lib/src/phy/utils/vector.c: In function ‘srslte_vec_max_abs_ci’:
/home/cewgroup/srsLTE/lib/src/phy/utils/vector.c:787:27: warning: passing argument 1 of ‘volk_32fc_index_max_16u’ from incompatible pointer type [-Wincompatible-pointer-types]
   volk_32fc_index_max_16u(&target,x,len);
                           ^
/home/cewgroup/srsLTE/lib/src/phy/utils/vector.c:787:27: note: expected ‘unsigned int *’ but argument is of type ‘uint16_t ** {aka short unsigned int **}’
/home/cewgroup/srsLTE/lib/src/phy/utils/vector.c:788:10: warning: return makes integer from pointer without a cast [-Wint-conversion]
   return target;

is there an incompatibility between volk libraries?

sirjansel0t, 2017-06-09:
I disabled the building srsLTE using volk and all tests passed.  This looks like it might be an issue with the volk libraries?

andrepuschmann, 2017-06-09:
Hey,
we've seen this issue before and its likely to be an issue between your volk built with an older gcc and the gcc currently installed on your system. Try recompiling volk or install from your package system.

suttonpd, 2017-06-09:
Having looked into this further, the issue is due to a volk version incompatibility. Specifically, v1.2 (the default on Ubuntu 16.04) uses: volk_32f_index_max_16u_generic(unsigned int* target, const float* src0, unsigned int num_points) while v1.3 (the latest version) uses: volk_32f_index_max_16u_generic(uint16_t* target, const float* src0, uint32_t num_points). You will need to upgrade the version of volk on your system to 1.3.

ismagom, 2017-06-09:
Hi again, we've fixed this issue just now. If you pull out the latest version from github the test should pass for you without having to upgrade volk.

closed 2017-06-09 by ismagom