srsran/srsRAN_4G #81: Analyze own file with pdsch_ue.c?

AndrewMcClelland, 2017-06-13:
Hey there,

I've successfully generated a signal file using ./pdsch_enodeb and demodulated it using ,/pdsch_ue. I would now like to be able to demodulate my own custom signal using ./pdsch_ue. To clarify, I have a binary file that contains a signal generated from separate software. I have tried to feed this input file into the ./pdsch_ue, however it is unable to decode the MIB. I have tried several different input methods (binary, normal ASCII) as well as altering some input flags, however it does not work. Can you please suggest another approach to doing this or if I'm missing something.

Additionally, is there anyway to display PSS/SSS/CRS info on the pdsch_ue side, after the signal is demodulated?

Thank you!

ismagom, 2017-06-13:
Hi Andrew, 

We recommend to use the srslte mailist (see README) for these kind of questions. This space should be reserved for bugs or issues. 

You should only use pdsch_ue with files generated with pdsch_enode or captured with usrp_capture_sync since it assumes fully time/frequency synchronized signal and 3/4 sampling rates instead of LTE standard sampling rates.

AndrewMcClelland, 2017-06-13:
Hi Ismael,

I apologize for posting here rather than in the maillist.

Thank you very much for your quick reply! Ok, that does make sense. However, do you know any easy way of modifying the pdsch_ue example to be able to handle custom generated LTE signal files? Or would there be a better way of going about this?

Ultimately, I have a modulated LTE signal that I am trying to perform EVM analysis on. My goal is to demodulate it, and perform EVM analysis on the original samples and the newly demodulated samples. I was hoping to use pdsch_ue to demodulate my custom signal, and then add on an EVM analysis aspect. If pdsch_ue is not ideal for this, do you have any suggestions on how to go about this? Using srsLTE or otherwise!

Sorry for the lengthy post and thank you very much in advance for your help!

Cheers

ismagom, 2017-06-13:
Hi Andrew, 

Yes the best is that you modify pdsch_ue for your custom signal. That's what it's intended for.

closed 2017-06-13 by AndrewMcClelland