srsran/srsRAN_4G #83: EPC that can work srsLTE

lrf8682004, 2017-06-15:
Thanks for the wonderful project. 
It seems to make the two parts srsue/srsENB work together and have a complete LTE system, EPC part is still missing. Could you suggest a EPC software to work with both srsue and srsENB? Is there any open source implementation available?

Thanks!

lrf8682004, 2017-06-15:
To make it more clearly,  I am only interested in the communication between UE and eNode. Is it possible to create the connection and communication between srsUE and srsENB without S1 interface(EPC)?

andrepuschmann, 2017-07-14:
Hey, I am afraid such a mode without S1 isn't possible with srsENB right now.

closed 2017-07-14 by andrepuschmann

FFY00, 2018-01-12:
@andrepuschmann could you stil suggest an open source EPC? Thanks in advance.

andrepuschmann, 2018-01-14:
Please check the mailing list for tips about EPCs. There has been a discussion about different alternatives just recently.