srsran/srsRAN_4G #85: Support for PCSC srsUE? (Hardware/external USIM)

Manouchehri, 2017-07-03:
Did the hard USIM support with PCSC by @pily112 (David Rupprecht) ever get uploaded/merged anywhere? Can't seem to find it.

http://www.softwareradiosystems.com/pipermail/srslte-users/2016-December/000617.html

andrepuschmann, 2017-07-04:
Hi, unfortunately this patch is not part of srsLTE due to licensing incompatibilities. I am not sure if it ever has been made publicly available.

closed 2017-07-04 by andrepuschmann

Manouchehri, 2017-07-04:
Was the licensing incompatibility due to PCSC, or just this particular
patch's code?

andrepuschmann, 2017-07-04:
I don't think there is any issue with PCSC. And to be correct it wasn't exactly an incompatibility with the patch either but rather an open issue with the licensing of the contributed code. For every pull request made to srsLTE we now ask the author to sign a Contributor License Agreement that handles such things in the future. That process is all done through github and should be straight forward.

Manouchehri, 2017-07-04:
Ah, did Rupprecht show you the patch privately? (I don't see any pull requests mentioning it.)

andrepuschmann, 2017-07-05:
That's right.