srsran/srsRAN_4G #92: uhd_tx_streamer_recv_async_msg String manipulation throws SIGABRT

kewinrausch, 2017-07-28:
While using version [242fe9a](https://github.com/kewinrausch/srsLTE/commit/242fe9adc93ec93b6366499bcf7a359a0d53607f) of the master branch, sometimes, when I leave the eNB open for some minutes where an UE is connected and registered with the Core Network (OpenAirInterface), the base station just crash with the following trace:

```
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_S_create

Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffd96fe700 (LWP 9861)]
0x00007ffff50f3067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56      ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff50f3067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff50f4448 in __GI_abort () at abort.c:89
#2  0x00007ffff59e0b3d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff59debb6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff59dec01 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff59dee19 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5a34ac6 in std::__throw_length_error(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff5a3eeb2 in std::string::_Rep::_S_create(unsigned long, unsigned long, std::allocator<char> const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff5a3ef14 in std::string::_M_mutate(unsigned long, unsigned long, unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff5f32403 in uhd_tx_streamer_recv_async_msg () from /usr/local/lib/libuhd.so.3
#10 0x00007ffff681669d in async_thread () from /home/empower/src/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so
#11 0x00007ffff7586064 in start_thread (arg=0x7fffd96fe700) at pthread_create.c:309
#12 0x00007ffff51a662d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

kewinrausch, 2017-08-03:
I've built srs software with -g flag in order to obtain more information in case of crash.
This way I obtained more feedback on the error, which you can see here:

```
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_S_create

Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffd8d1c700 (LWP 3680)]
0x00007ffff50f2067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56      ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff50f2067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff50f3448 in __GI_abort () at abort.c:89
#2  0x00007ffff59dfb3d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff59ddbb6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff59ddc01 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff59dde19 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5a33ac6 in std::__throw_length_error(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff5a3deb2 in std::string::_Rep::_S_create(unsigned long, unsigned long, std::allocator<char> const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff5a3df14 in std::string::_M_mutate(unsigned long, unsigned long, unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff5f3ddcc in uhd_tx_streamer_send () from /usr/local/lib/libuhd.so.3
#10 0x00007ffff6816941 in rf_uhd_send_timed (h=0x509ce50, data=0x50f9400, nsamples=5760, secs=<optimized out>, frac_secs=<optimized out>,
    has_time_spec=<optimized out>, blocking=<optimized out>, is_start_of_burst=false, is_end_of_burst=false)
    at /home/empower/src/srsLTE/lib/src/phy/rf/rf_uhd_imp.c:681
#11 0x00007ffff6815601 in srslte_rf_send_timed2 (rf=<optimized out>, data=<optimized out>, nsamples=<optimized out>,
    secs=<optimized out>, frac_secs=<optimized out>, is_start_of_burst=<optimized out>, is_end_of_burst=<optimized out>)
    at /home/empower/src/srsLTE/lib/src/phy/rf/rf_imp.c:332
#12 0x00000000005636ce in srslte::radio::tx (this=0x7fffde6da018, buffer=buffer@entry=0x50f9400, nof_samples=nof_samples@entry=5760,
    tx_time=...) at /home/empower/src/srsLTE/lib/src/radio/radio.cc:194
#13 0x00000000004e18e3 in srsenb::phch_common::worker_end (this=0x7fffed13a350, tx_mutex_cnt=2, buffer=0x50f9400, nof_samples=5760,
    tx_time=...) at /home/empower/src/srsLTE/srsenb/src/phy/phch_common.cc:88
#14 0x00000000004dfe9e in srsenb::phch_worker::work_imp (this=0x7fffde316010)
    at /home/empower/src/srsLTE/srsenb/src/phy/phch_worker.cc:297
#15 0x00000000004e7b09 in srslte::thread_pool::worker::run_thread (this=0x7fffde316010)
    at /home/empower/src/srsLTE/lib/src/common/thread_pool.cc:61
#16 0x00000000004b1369 in thread::thread_function_entry (_this=<optimized out>)
    at /home/empower/src/srsLTE/lib/include/srslte/common/threads.h:73
#17 0x00007ffff7585064 in start_thread (arg=0x7fffd8d1c700) at pthread_create.c:309
#18 0x00007ffff51a562d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

Just let me know if I have to add more output to attach to the issue (and which sort of data you will need).

ismagom, 2017-08-18:
Do you still see it in recent versions?

kewinrausch, 2017-09-04:
I'll update my version in the these days and let you know...

liuzhikui, 2017-11-24:
Hello, I have met same issue with you. Can you tell me how to solve it?

kewinrausch, 2017-11-24:
Actually I did not solve it; I just updated my version of srsLTE to the last version and this issue become less frequent. I'm not sure if it has been fixed by other updates thus...

liuzhikui, 2017-11-24:
ismagom tell me that this is an issue with USRP or USB driver,but I feel a lot confounded.

kewinrausch, 2017-12-05:
Just had again this same error.

ismagom, 2018-01-20:
This should be fixed in the most recent release. I will close it, please reopen if it's still happening.

closed 2018-01-20 by ismagom