srsran/srsRAN_4G #94: Error during UE attachment

kewinrausch, 2017-07-31:
Branch: Master
Commit: [242fe9a](https://github.com/kewinrausch/srsLTE/commit/242fe9adc93ec93b6366499bcf7a359a0d53607f)

This error occurs just after the RACH procedures, and appears randomly (it's not very common).

```
Setting Sampling frequency 5.76 MHz
Setting TX/RX offset 93 samples, 16.15 us

==== eNodeB started ===
Type <t> to view trace
RACH:  tti=8091, preamble=10, offset=0, temp_crnti=0x46

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd8d10700 (LWP 11065)]
0x00000000005d9b1a in _mm256_load_si256 (__P=0x7fffde328390) at /usr/lib/gcc/x86_64-linux-gnu/4.9/include/avxintrin.h:880
880       return *__P;
(gdb) bt
#0  0x00000000005d9b1a in _mm256_load_si256 (__P=0x7fffde328390) at /usr/lib/gcc/x86_64-linux-gnu/4.9/include/avxintrin.h:880
#1  srslte_vec_dot_prod_sss_avx2 (x=0x7fffde328390, y=0x7fffd8d0f700, len=20)
    at /home/empower/src/srsLTE/lib/src/phy/utils/vector_simd.c:105
#2  0x00000000005ade83 in srslte_vec_dot_prod_sss (x=0x7fffde328390, y=0x7fffd8d0f700, len=20)
    at /home/empower/src/srsLTE/lib/src/phy/utils/vector.c:665
#3  0x00000000005d11fb in srslte_uci_decode_cqi_pucch (q=0x7fffde328190, b_bits=0x7fffd8d0f700, cqi_data=0x7fffd8d0fbb0 "", cqi_len=4)
    at /home/empower/src/srsLTE/lib/src/phy/phch/uci.c:152
#4  0x000000000059e399 in srslte_pucch_decode (q=0x7fffde328128, format=SRSLTE_PUCCH_FORMAT_2, n_pucch=0, sf_idx=0, rnti=70,
    sf_symbols=0x5e38900, ce=0x5e40d00, noise_estimate=0.00436232565, bits=0x7fffd8d0fbb0 "")
    at /home/empower/src/srsLTE/lib/src/phy/phch/pucch.c:810
#5  0x00000000005ba0de in get_pucch (q=0x7fffde318fc8, rnti=70, pdcch_n_cce=0, sf_rx=0, uci_data=0x7fffd8d0fc70, bits=0x7fffd8d0fbb0 "")
    at /home/empower/src/srsLTE/lib/src/phy/enb/enb_ul.c:238
#6  0x00000000005ba17c in srslte_enb_ul_get_pucch (q=0x7fffde318fc8, rnti=70, pdcch_n_cce=0, sf_rx=0, uci_data=0x7fffd8d0fc70)
    at /home/empower/src/srsLTE/lib/src/phy/enb/enb_ul.c:254
#7  0x000000000057fb78 in srsenb::phch_worker::decode_pucch (this=0x7fffde30a010, tti_rx=8120)
    at /home/empower/src/srsLTE/srsenb/src/phy/phch_worker.cc:497
#8  0x000000000057e1b2 in srsenb::phch_worker::work_imp (this=0x7fffde30a010)
    at /home/empower/src/srsLTE/srsenb/src/phy/phch_worker.cc:252
#9  0x000000000058a439 in srslte::thread_pool::worker::run_thread (this=0x7fffde30a010)
    at /home/empower/src/srsLTE/lib/src/common/thread_pool.cc:61
#10 0x00000000005484e5 in thread::thread_function_entry (_this=0x7fffde30a010)
    at /home/empower/src/srsLTE/lib/include/srslte/common/threads.h:73
#11 0x00007ffff7586064 in start_thread (arg=0x7fffd8d10700) at pthread_create.c:309
#12 0x00007ffff519962d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

kewinrausch, 2017-07-31:
Additional information about this one: the error seems to be generated when the project is compiled with the "Debug" profile, like including the the CMakeLists.txt the following line: `set(CMAKE_BUILD_TYPE "Debug")`

ismagom, 2017-08-22:
Is this issue still active? I can not reproduce it in Debug mode in commit 40273e4a using srsENB

kewinrausch, 2017-09-04:
I just tried with [a631798](https://github.com/srsLTE/srsLTE/commit/a6317980579d29a48ccbc7047e3ec38f8bf4cd4c) and I still get the error, as you can see here, by using GDB.

Could it be that this problem arise with certain version of the needed libraries, for example?

- GCC is version 4.9
- I have fftw-3.3.6-pl2
- AVX is supported as i can see from the CPU flags:

```
???@???:~/src$ cat /proc/cpuinfo | grep avx
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm ida arat epb xsaveopt pln pts dtherm tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid
```

**Error trace:**
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffde94a700 (LWP 4324)]
[New Thread 0x7fffde149700 (LWP 4325)]
[New Thread 0x7fffdd948700 (LWP 4326)]
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file ./enb.conf...
[New Thread 0x7fffdd147700 (LWP 4327)]
[New Thread 0x7fffdc946700 (LWP 4328)]
[New Thread 0x7fffdc145700 (LWP 4329)]
[Thread 0x7fffdc145700 (LWP 4329) exited]
[Thread 0x7fffdc946700 (LWP 4328) exited]
[New Thread 0x7fffdc946700 (LWP 4330)]
[INFO] [UHDlinux; GNU C++ version 4.9.2; Boost_105500; UHD_3.11.0.git-208-g1da86f9c]
[New Thread 0x7fffdc145700 (LWP 4331)]
[New Thread 0x7fffdb944700 (LWP 4332)]
[New Thread 0x7fffdb143700 (LWP 4333)]
[Thread 0x7fffdb143700 (LWP 4333) exited]
[Thread 0x7fffdb944700 (LWP 4332) exited]
[New Thread 0x7fffdb944700 (LWP 4334)]
[New Thread 0x7fffdb143700 (LWP 4335)]
[Thread 0x7fffdb143700 (LWP 4335) exited]
[Thread 0x7fffdb944700 (LWP 4334) exited]
[New Thread 0x7fffdb944700 (LWP 4336)]
[New Thread 0x7fffdb143700 (LWP 4337)]
[Thread 0x7fffdb143700 (LWP 4337) exited]
[Thread 0x7fffdb944700 (LWP 4336) exited]
Opening USRP with args: type=b200,master_clock_rate=30.72e6
[New Thread 0x7fffdb944700 (LWP 4338)]
[New Thread 0x7fffdb143700 (LWP 4339)]
[Thread 0x7fffdb143700 (LWP 4339) exited]
[Thread 0x7fffdb944700 (LWP 4338) exited]
[New Thread 0x7fffdb944700 (LWP 4340)]
[New Thread 0x7fffdb143700 (LWP 4341)]
[Thread 0x7fffdb143700 (LWP 4341) exited]
[Thread 0x7fffdb944700 (LWP 4340) exited]
[New Thread 0x7fffdb944700 (LWP 4342)]
[New Thread 0x7fffdb143700 (LWP 4343)]
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[New Thread 0x7fffda533700 (LWP 4344)]
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] pass
[INFO] [B200] Performing register loopback test...
[INFO] [B200] pass
[INFO] [AD936X] Performing CODEC loopback test...
[INFO] [AD936X] CODEC loopback test passed
[INFO] [AD936X] Performing CODEC loopback test...
[INFO] [AD936X] CODEC loopback test passed
[INFO] [B200] Asking for clock rate 30.720000 MHz...
[INFO] [B200] Actually got clock rate 30.720000 MHz.
[INFO] [CORES] Performing timer loopback test...
[... 33 lines cut ...]
#0  0x00000000005cd516 in _mm256_load_si256 (__P=0x7fffde969390) at /usr/lib/gcc/x86_64-linux-gnu/4.9/include/avxintrin.h:880
#1  srslte_vec_dot_prod_sss_avx2 (x=0x7fffde969390, y=0x7fffd9350700, len=20)
    at /home/empower/src/srsLTE/lib/src/phy/utils/vector_simd.c:105
#2  0x00000000005a187f in srslte_vec_dot_prod_sss (x=0x7fffde969390, y=0x7fffd9350700, len=20)
    at /home/empower/src/srsLTE/lib/src/phy/utils/vector.c:665
#3  0x00000000005c4bf7 in srslte_uci_decode_cqi_pucch (q=0x7fffde969190, b_bits=0x7fffd9350700, cqi_data=0x7fffd9350bb0 "", cqi_len=4)
    at /home/empower/src/srsLTE/lib/src/phy/phch/uci.c:152
#4  0x0000000000591c7a in srslte_pucch_decode (q=0x7fffde969128, format=SRSLTE_PUCCH_FORMAT_2, n_pucch=0, sf_idx=0, rnti=70,
    sf_symbols=0x5e18200, ce=0x5e20600, noise_estimate=0.00617822818, bits=0x7fffd9350bb0 "")
    at /home/empower/src/srsLTE/lib/src/phy/phch/pucch.c:812
#5  0x00000000005adada in get_pucch (q=0x7fffde959fc8, rnti=70, pdcch_n_cce=0, sf_rx=0, uci_data=0x7fffd9350c70, bits=0x7fffd9350bb0 "")
    at /home/empower/src/srsLTE/lib/src/phy/enb/enb_ul.c:238
#6  0x00000000005adb78 in srslte_enb_ul_get_pucch (q=0x7fffde959fc8, rnti=70, pdcch_n_cce=0, sf_rx=0, uci_data=0x7fffd9350c70)
    at /home/empower/src/srsLTE/lib/src/phy/enb/enb_ul.c:254
#7  0x00000000005733de in srsenb::phch_worker::decode_pucch (this=0x7fffde94b010, tti_rx=5840)
    at /home/empower/src/srsLTE/srsenb/src/phy/phch_worker.cc:497
#8  0x0000000000571a18 in srsenb::phch_worker::work_imp (this=0x7fffde94b010)
    at /home/empower/src/srsLTE/srsenb/src/phy/phch_worker.cc:252
#9  0x000000000057dc9f in srslte::thread_pool::worker::run_thread (this=0x7fffde94b010)
    at /home/empower/src/srsLTE/lib/src/common/thread_pool.cc:61
#10 0x000000000053d067 in thread::thread_function_entry (_this=0x7fffde94b010)
    at /home/empower/src/srsLTE/lib/include/srslte/common/threads.h:73
#11 0x00007ffff7bc7064 in start_thread (arg=0x7fffd9351700) at pthread_create.c:309
#12 0x00007ffff57da62d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

commit 2dbc0f06 (2017-09-05): Fix memory alignment in PUCCH processing. Fixes #94
closed 2017-09-05 by ismagom