srsran/srsRAN_4G #96: prach: Repeated offset miss with certain UE devices

ttsou, 2017-08-17:
Huawei E398 USB modem (Qualcomm MDM9200) shows consistent and repeated PRACH measurement offset. Behavior may be device/chipset specific; identical measurements are obtained with two separate modems of the same model.

Expected PRACH offset is seen with a Google Nexus 4. Errant timing behavior not seen with Amarisoft eNode.

**Hauwei E398 (Qualcomm MDM9200)**
```
15:04:47.821276 [PHY0] Info    [01340] PRACH: 0/1, preamble=41, offset=87.7 us, peak2avg=834.0, max_offset=30.0 us
15:04:47.841231 [PHY0] Info    [01360] PRACH: 0/1, preamble=32, offset=87.7 us, peak2avg=833.9, max_offset=30.0 us
15:04:47.861188 [PHY0] Info    [01380] PRACH: 0/1, preamble=47, offset=87.7 us, peak2avg=833.7, max_offset=30.0 us
15:04:47.881238 [PHY0] Info    [01400] PRACH: 0/1, preamble=39, offset=87.7 us, peak2avg=834.6, max_offset=30.0 us
15:04:47.901235 [PHY0] Info    [01420] PRACH: 0/1, preamble=13, offset=87.7 us, peak2avg=834.2, max_offset=30.0 us
15:04:47.921252 [PHY0] Info    [01440] PRACH: 0/1, preamble=41, offset=87.7 us, peak2avg=830.0, max_offset=30.0 us
15:04:47.941208 [PHY0] Info    [01460] PRACH: 0/1, preamble=51, offset=87.7 us, peak2avg=835.3, max_offset=30.0 us
15:04:47.961266 [PHY0] Info    [01480] PRACH: 0/1, preamble=34, offset=87.7 us, peak2avg=831.1, max_offset=30.0 us
15:04:47.981220 [PHY0] Info    [01500] PRACH: 0/1, preamble=35, offset=87.7 us, peak2avg=830.6, max_offset=30.0 us
15:04:48.001275 [PHY0] Info    [01520] PRACH: 0/1, preamble=22, offset=87.7 us, peak2avg=831.4, max_offset=30.0 us
15:04:48.021235 [PHY0] Info    [01540] PRACH: 0/1, preamble=48, offset=87.7 us, peak2avg=831.2, max_offset=30.0 us
15:04:48.041232 [PHY0] Info    [01560] PRACH: 0/1, preamble=25, offset=87.7 us, peak2avg=824.6, max_offset=30.0 us
15:04:48.061252 [PHY0] Info    [01580] PRACH: 0/1, preamble=17, offset=87.7 us, peak2avg=824.7, max_offset=30.0 us
15:04:48.101259 [PHY0] Info    [01620] PRACH: 0/1, preamble=14, offset=87.7 us, peak2avg=825.1, max_offset=30.0 us
15:04:48.121213 [PHY0] Info    [01640] PRACH: 0/1, preamble=35, offset=87.7 us, peak2avg=824.1, max_offset=30.0 us
```
**Google Nexus 4 (Qualcomm APQ8064)**
```
15:20:35.484196 [PHY0] Info    [09721] PRACH: 0/1, preamble=19, offset=0.0 us, peak2avg=692.6, max_offset=30.0 us
15:20:35.486039 [MAC ] Info    [09722] SCHED: Added user rnti=0x46
15:20:35.486057 [MAC ] Info    [09722] SCHED: Set bearer config lc_id=0, direction=3
15:20:35.486111 [RRC ] Info    Added new user rnti=0x46
15:20:35.486117 [MAC ] Info    [09723] RACH:  tti=9721, preamble=19, offset=0, temp_crnti=0x46
15:20:35.487045 [MAC ] Info    [09724] SCHED: RAR, ra_id=0, rnti=0x46, rarnti_idx=2, start_rb=0, n_rb=3, rar_grant_rba=202, rar_grant_mcs=0
15:20:35.487073 [PHY1] Info    [09724] PDCCH: DL DCI Format1A rnti=0x2, cce_index=0, L=4, tti_tx=9728
15:20:35.487082 [PHY1] Info    [09724] PDSCH: rnti=0x2, l_crb= 3, rb_start=0, harq=0, tbs=7, mcs=0, rv=0, tti_tx=9728
             0000: 53 00 01 94 0c 00 46 
15:20:35.489073 [MAC ] Info    [09726] SCHED: RAR tx rnti=0x46, pid=6, dci=0,0, grant=2,3, n_rtx=0, tbs=7, bsr=0 (0)
15:20:35.497103 [PHY1] Info    [09734] PUSCH: rnti=0x46, prb=(2,5), tbs=7, mcs=0, rv=0, snr=20.9 dB, n_iter=1, crc=OK, dec_time=  37 us
             0000: 00 58 0d 3b 14 dc 96 
15:20:35.497132 [PHY1] Info    [09734] PHICH: rnti=0x46, hi=1, I_lowest=2, n_dmrs=0, tti_tx=9738
15:20:35.497322 [PDCP] Info    RX SRB0 PDU
             0000: 58 0d 3b 14 dc 96 
15:20:35.497362 [RRC ] Info    Rx SRB0 PDU
             0000: 58 0d 3b 14 dc 96 
15:20:35.497373 [RRC ] Info    SRB0 - Rx: RRC Connection Request
             0000: 58 0d 3b 14 dc 96 
15:20:35.497380 [RRC ] Info    Allocated SR resources for time-frequency slot (0, 0), N_pucch_sr=0, I_sr=16
15:20:35.497384 [RRC ] Info    Allocated CQI resources for time-frequency slot (0, 0), n_pucch_2=0, pmi_cfg_idx=37
15:20:35.497387 [MAC ] Info    [09734] Registering rnti=0x46 to PHY...
15:20:35.510933 [MAC ] Info    [09747] Done registering rnti=0x46 to PHY...
```

**Configuration**
```
Amarisoft MME (2016-10-13)
SRS eNodeB (00be4a5de7840)
SRS default configuration (ex. dl_earfcn=300, tx_gain = 0)
RF isolated environment (>90 dB @ 2 GHz)
UE - Huawei E398, LG Nexus 4
```

ismagom, 2017-08-20:
Thanks for your input.

I've tried Huawei dongle E3276 and works perfectly, PRACH detected with offset=0. 

In Amarisoft do you see exactly the same offset with both modems? 

Can you try these please?: 

1) Set expert.max_prach_offset_us=90 in enb.conf

2) Set rf.time_adv_nsamples to a value slightly lower than the value hardcoded for your hardware and sampling frequency in function radio::set_tx_rate() in lib/src/radio/radio.cc. For example for b200, 5.76 MHz set rf.time_adv_nsamples=85 or 80 (instead of 93, line 330). We might be compensating too much TX/RX delay and the PRACH is "arriving" with a negative time offset. 

Thank you

ttsou, 2017-08-21:
@ismagom wrote:
>    Set expert.max_prach_offset_us=90 in enb.conf

The trace proceeds with the added offset allowance, but the connection procedure is still incomplete with timeout of RRCConnectionSetup. No S1AP/NAS or other eNB-EPC activity after startup.

```
15:57:47.737273 [PHY0] Info    [00000] Worker 0 configured cell 100 PRB
15:57:47.845555 [PHY1] Info    [00000] Worker 0 configured cell 100 PRB
15:57:49.120770 [MAC ] Info    [00000] Resetting MAC
15:57:49.308523 [S1AP] Info    Connecting to MME 127.0.1.100:36412
15:57:49.308662 [S1AP] Info    SCTP socket established with MME
15:57:49.308686 [S1AP] Info    Sending s1SetupRequest
             0000: 00 11 00 2d 00 00 04 00 3b 00 08 00 00 f1 10 00 
             0010: 00 00 10 00 3c 40 0a 03 80 73 72 73 65 6e 62 30 
15:57:49.308840 [S1AP] Info    Received S1AP PDU
             0000: 20 11 00 17 00 00 02 00 69 00 0b 00 00 00 f1 10 
             0010: 00 00 80 01 00 01 00 57 40 01 32 
15:57:49.308865 [S1AP] Info    Received S1SetupResponse
15:57:49.572788 [PHY0] Info    [00000] Starting RX/TX thread nof_prb=100, sf_len=23040
15:57:50.819200 [PHY0] Info    [00740] PRACH: 0/1, preamble=48, offset=87.7 us, peak2avg=183.9, max_offset=90.0 us
15:57:50.820957 [MAC ] Info    [00742] SCHED: Added user rnti=0x46
15:57:50.820968 [MAC ] Info    [00742] SCHED: Set bearer config lc_id=0, direction=3
15:57:50.821015 [RRC ] Info    Added new user rnti=0x46
15:57:50.821020 [MAC ] Info    [00742] RACH:  tti=741, preamble=48, offset=87, temp_crnti=0x46
15:57:50.821068 [MAC ] Info    [00743] SCHED: RAR, ra_id=0, rnti=0x46, rarnti_idx=2, start_rb=0, n_rb=3, rar_grant_rba=202, rar_grant_mcs=0
15:57:50.821099 [PHY1] Info    [00743] PDCCH: DL DCI Format1A rnti=0x2, cce_index=0, L=4, tti_tx=747
15:57:50.821107 [PHY1] Info    [00743] PDSCH: rnti=0x2, l_crb= 3, rb_start=0, harq=0, tbs=7, mcs=0, rv=0, tti_tx=747
             0000: 70 05 71 94 0c 00 46 
15:57:50.823110 [MAC ] Info    [00745] SCHED: RAR tx rnti=0x46, pid=1, dci=0,0, grant=2,3, n_rtx=0, tbs=7, bsr=0 (0)
15:57:50.831143 [PHY1] Info    [00753] PUSCH: rnti=0x46, prb=(2,5), tbs=7, mcs=0, rv=0, snr=-0.7 dB, n_iter=4, crc=KO, dec_time=  38 us
             0000: 97 55 c6 4a d5 47 6e 
15:57:50.831159 [MAC ] Info    [00753] SCHED: UL retx rnti=0x46, pid=1, dci=0,0, grant=2,3, n_rtx=1, tbs=7, bsr=0 (0)
15:57:50.831178 [PHY1] Info    [00753] PHICH: rnti=0x46, hi=0, I_lowest=2, n_dmrs=0, tti_tx=757
15:57:50.839126 [PHY1] Info    [00761] PUSCH: rnti=0x46, prb=(2,5), tbs=7, mcs=0, rv=2, snr=0.6 dB, n_iter=4, crc=KO, dec_time=  33 us
             0000: 9f a5 70 e1 2f ba 69 
15:57:50.839145 [MAC ] Info    [00761] SCHED: UL retx rnti=0x46, pid=1, dci=0,0, grant=2,3, n_rtx=2, tbs=7, bsr=0 (0)
15:57:50.839172 [PHY1] Info    [00761] PHICH: rnti=0x46, hi=0, I_lowest=2, n_dmrs=0, tti_tx=765
15:57:50.839232 [PHY0] Info    [00760] PRACH: 0/2, preamble=27, offset=76.3 us, peak2avg=73.9, max_offset=90.0 us
15:57:50.840978 [MAC ] Info    [00762] SCHED: Added user rnti=0x47
15:57:50.840987 [MAC ] Info    [00762] SCHED: Set bearer config lc_id=0, direction=3
15:57:50.841048 [RRC ] Info    Added new user rnti=0x47
15:57:50.841053 [MAC ] Info    [00762] RACH:  tti=761, preamble=27, offset=76, temp_crnti=0x47
15:57:50.841073 [PHY0] Info    [00762] PRACH: 1/2, preamble=28, offset=87.7 us, peak2avg=368.8, max_offset=90.0 us
15:57:50.841101 [MAC ] Info    [00763] SCHED: RAR, ra_id=0, rnti=0x47, rarnti_idx=2, start_rb=0, n_rb=3, rar_grant_rba=202, rar_grant_mcs=0
15:57:50.841126 [PHY1] Info    [00763] PDCCH: DL DCI Format1A rnti=0x2, cce_index=0, L=4, tti_tx=767
15:57:50.841131 [PHY1] Info    [00763] PDSCH: rnti=0x2, l_crb= 3, rb_start=0, harq=0, tbs=7, mcs=0, rv=0, tti_tx=767
             0000: 5b 04 c1 94 0c 00 47 
15:57:50.842759 [MAC ] Info    [00764] SCHED: Added user rnti=0x48
15:57:50.842765 [MAC ] Info    [00764] SCHED: Set bearer config lc_id=0, direction=3
15:57:50.842824 [RRC ] Info    Added new user rnti=0x48
15:57:50.842829 [MAC ] Info    [00764] RACH:  tti=761, preamble=28, offset=87, temp_crnti=0x48
15:57:50.843061 [MAC ] Info    [00765] SCHED: RAR, ra_id=1, rnti=0x48, rarnti_idx=2, start_rb=0, n_rb=3, rar_grant_rba=202, rar_grant_mcs=0
15:57:50.843085 [MAC ] Info    [00765] SCHED: RAR tx rnti=0x47, pid=5, dci=0,0, grant=2,3, n_rtx=0, tbs=7, bsr=0 (0)
15:57:50.843113 [PHY1] Info    [00765] PDCCH: DL DCI Format1A rnti=0x2, cce_index=0, L=4, tti_tx=769
15:57:50.843118 [PHY1] Info    [00765] PDSCH: rnti=0x2, l_crb= 3, rb_start=0, harq=0, tbs=7, mcs=0, rv=0, tti_tx=769
             0000: 5c 05 71 94 0c 00 48 
15:57:50.845024 [MAC ] Info    [00767] SCHED: RAR tx rnti=0x48, pid=7, dci=0,0, grant=2,3, n_rtx=0, tbs=7, bsr=0 (0)
15:57:50.847102 [PHY1] Info    [00769] PUSCH: rnti=0x46, prb=(2,5), tbs=7, mcs=0, rv=3, snr=0.1 dB, n_iter=4, crc=KO, dec_time=  38 us
             0000: 63 68 59 72 3a 00 bc 
15:57:50.847119 [MAC ] Info    [00769] SCHED: UL retx rnti=0x46, pid=1, dci=0,0, grant=2,3, n_rtx=3, tbs=7, bsr=0 (0)
15:57:50.847145 [PHY1] Info    [00769] PHICH: rnti=0x46, hi=0, I_lowest=2, n_dmrs=0, tti_tx=773
15:57:50.851121 [PHY1] Info    [00773] PUSCH: rnti=0x47, prb=(2,5), tbs=7, mcs=0, rv=0, snr=-1.1 dB, n_iter=4, crc=KO, dec_time=  43 us
             0000: 9b 1f d9 ac 8b 09 48 
15:57:50.851150 [MAC ] Info    [00773] SCHED: UL retx rnti=0x47, pid=5, dci=0,0, grant=2,3, n_rtx=1, tbs=7, bsr=0 (0)
15:57:50.851179 [PHY1] Info    [00773] PHICH: rnti=0x47, hi=0, I_lowest=2, n_dmrs=0, tti_tx=777
[... cut ...]
15:57:51.333116 [PHY1] Info    [01255] PHICH: rnti=0x53, hi=0, I_lowest=2, n_dmrs=0, tti_tx=1259
15:57:51.335870 [RRC ] Warning User rnti=0x53 expired RRCConnectionSetup deadline: 0:45062>0:40000 us
15:57:51.335880 [RRC ] Info    User rnti=0x53 timed out. Exists in s1ap=no
15:57:51.335888 [RRC ] Info    Disconnecting rnti=0x53.
15:57:51.335893 [PHY0] Error   [01256] Removing user: rnti=0x53 does not exist
15:57:51.335919 [MAC ] Info    [01257] User rnti=0x53 removed from MAC/PHY
15:57:51.335937 [RRC ] Info    Removed user rnti=0x53
```

ttsou, 2017-08-21:
@ismagom wrote:
>Set rf.time_adv_nsamples to a value slightly lower than the value hardcoded for your hardware and sampling frequency in function radio::set_tx_rate() in lib/src/radio/radio.cc. For example for b200, 5.76 MHz set rf.time_adv_nsamples=85 or 80 (instead of 93, line 330).

Setting 'time_adv_nsamples = 150' from the automatic 20 MHz value of 175 seems to resolve the issue. Huawei E398 attachment is now stable and showing similar trace behaviour as other devices in #97.

ismagom, 2017-08-22:
Thanks for your help, just fixed it. 

Have you had the chance to test the same dongle in other BW configurations?

commit 5e41ea9d (2017-08-22): Fixes #96. Reduce TX/RX offset for 20MHz.
closed 2017-08-22 by ismagom