srsran/srsRAN_4G #97: nas-eps: UE does not complete EPS default bearer process

ttsou, 2017-08-17:
Using LG Nexus 4 and Amarisoft MME, UE connection process proceeds normally until repeatedly stalling with the following condition.

```
PDN connectivity reject (Multiple PDN connections for a given APN not allowed)
```

**Full S1AP/NAS-EPS Log**
```
    7 29.751689845    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 202 id-initialUEMessage, Attach request, PDN connectivity request
    8 29.751770322  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 106 SACK id-downlinkNASTransport, Identity request
    9 29.771554963    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 142 SACK id-uplinkNASTransport, Identity response
   10 29.771639878  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 138 SACK id-downlinkNASTransport, Authentication request
   11 29.871512916    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 142 SACK id-uplinkNASTransport, Authentication response
   12 29.871576443  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 122 SACK id-downlinkNASTransport, Security mode command
   13 29.931497711    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 146 SACK id-uplinkNASTransport, Security mode complete
   14 29.931568396  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 114 SACK id-downlinkNASTransport, ESM information request
   15 30.051516408    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 142 SACK id-uplinkNASTransport, ESM information response
   16 30.051600702  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 266 SACK id-InitialContextSetup, InitialContextSetupRequest , Attach accept, Activate default EPS bearer context request (PDN type IPv4 only allowed)
   17 30.071478100    127.0.1.1 → 127.0.1.100  S1AP 122 SACK id-InitialContextSetup, InitialContextSetupResponse
   19 30.272029939    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 122 id-uplinkNASTransport, Attach complete, Activate default EPS bearer context accept
   20 30.272093163  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 154 SACK id-downlinkNASTransport, EMM information
   22 30.571510379    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 154 id-uplinkNASTransport, PDN connectivity request
   23 30.571579008  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 162 SACK id-E-RABSetup, E_RABSetupRequest , Activate default EPS bearer context request
   24 30.571685585    127.0.1.1 → 127.0.1.100  S1AP 118 SACK id-E-RABSetup, E_RABSetupResponse
   26 30.772020495    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 118 id-uplinkNASTransport, Activate default EPS bearer context accept
   36 60.656330912    127.0.1.1 → 127.0.1.100  S1AP 90 id-UEContextReleaseRequest, UEContextReleaseRequest  [RadioNetwork-cause=user-inactivity]
   37 60.656376556  127.0.1.100 → 127.0.1.1    S1AP 102 SACK id-UEContextRelease, UEContextReleaseCommand  [RadioNetwork-cause=user-inactivity]
   38 60.656452348    127.0.1.1 → 127.0.1.100  S1AP 98 SACK id-UEContextRelease, UEContextReleaseComplete
  114 187.931756673    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 178 id-initialUEMessage, Tracking area update request
  115 187.931819806  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 154 SACK id-downlinkNASTransport, Tracking area update accept
  116 187.971603431    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 134 SACK id-uplinkNASTransport, Tracking area update complete
  117 187.971660830  127.0.1.100 → 127.0.1.1    S1AP 98 SACK id-UEContextRelease, UEContextReleaseCommand   [NAS-cause=normal-release]
  118 187.971717378    127.0.1.1 → 127.0.1.100  S1AP 98 SACK id-UEContextRelease, UEContextReleaseComplete
  124 190.351740945    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 178 id-initialUEMessage, Tracking area update request
  125 190.351811978  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 154 SACK id-downlinkNASTransport, Tracking area update accept
  126 190.411542694    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 134 SACK id-uplinkNASTransport, Tracking area update complete
  127 190.411597537  127.0.1.100 → 127.0.1.1    S1AP 98 SACK id-UEContextRelease, UEContextReleaseCommand   [NAS-cause=normal-release]
  128 190.411644164    127.0.1.1 → 127.0.1.100  S1AP 98 SACK id-UEContextRelease, UEContextReleaseComplete
  134 192.691617492    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 178 id-initialUEMessage, Tracking area update request
  135 192.691692391  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 154 SACK id-downlinkNASTransport, Tracking area update accept
  136 192.711495910    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 170 SACK id-uplinkNASTransport, PDN connectivity request
  137 192.711555872  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 114 SACK id-downlinkNASTransport, PDN connectivity reject (Multiple PDN connections for a given APN not allowed)
  138 192.711568682    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 134 SACK id-uplinkNASTransport, Tracking area update complete
  139 192.711589562  127.0.1.100 → 127.0.1.1    S1AP 98 SACK id-UEContextRelease, UEContextReleaseCommand   [NAS-cause=normal-release]
  140 192.711729001    127.0.1.1 → 127.0.1.100  S1AP 98 SACK id-UEContextRelease, UEContextReleaseComplete
  146 194.851609716    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 122 id-initialUEMessage, Service request
  147 194.851690878  127.0.1.100 → 127.0.1.1    S1AP 198 SACK id-InitialContextSetup, InitialContextSetupRequest
  148 195.031504848    127.0.1.1 → 127.0.1.100  S1AP 138 SACK id-InitialContextSetup, InitialContextSetupResponse
  154 195.232023893    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 154 id-uplinkNASTransport, PDN connectivity request
  155 195.232091923  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 114 SACK id-downlinkNASTransport, PDN connectivity reject (Multiple PDN connections for a given APN not allowed)
  157 215.431158352    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 154 id-uplinkNASTransport, PDN connectivity request
  158 215.431216357  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 114 SACK id-downlinkNASTransport, PDN connectivity reject (Multiple PDN connections for a given APN not allowed)
  160 223.363090246    127.0.1.1 → 127.0.1.100  S1AP/NAS-EPS 154 id-uplinkNASTransport, PDN connectivity request
  161 223.363161814  127.0.1.100 → 127.0.1.1    S1AP/NAS-EPS 114 SACK id-downlinkNASTransport, PDN connectivity reject (Multiple PDN connections for a given APN not allowed)
```

**Configuration**
```
Amarisoft MME (2016-10-13)
SRS eNodeB (00be4a5de7840)
SRS default configuration (ex. dl_earfcn=300, tx_gain = 0)
RF isolated environment (>90 dB @ 2 GHz)
UE - LG Nexus 4
```

ttsou, 2017-08-22:
Issue is (obviously) related to APN. Switching UE from secondary APN to default APN (Amarisoft MME default "test123") removes the PDN connection reject. Why the error occurs with SRS but not Amarisoft eNB remains unknown.

andrepuschmann, 2018-04-03:
Hey Tom, is this still an issue with the latest release?

andrepuschmann, 2018-06-25:
Closing for now. Please reopen if needed.

closed 2018-06-25 by andrepuschmann