srsran/srsRAN_4G #98: Can this project be extended to implement NB-IoT?

desword, 2017-08-18:
Can this project be extended to implement NB-IoT on USRP N210?

https://arxiv.org/pdf/1705.03529.pdf

I find that there is a paper describe this project can be used to implement NB-IoT.
However, I can not find the introduced component like nbiot_call_search module in example applications.

ismagom, 2017-08-18:
Hi,

NB-IoT extensions are currently not available under the open source version. Please contact  sales@softwareradiosystems.com for more information.

closed 2017-08-18 by ismagom

PhatHub, 2021-08-05:
@ismagom 
Hi! Are NB-IoT extensions still not available? I see that there are example UE and enodeb ... do those examples allow for running an NBIoT eNodeB testbed? Or does that still require the non-Open Source version?
Thanks!

> Hi,
> 
> NB-IoT extensions are currently not available under the open source version. Please contact [sales@softwareradiosystems.com](mailto:sales@softwareradiosystems.com) for more information.

andrepuschmann, 2021-08-07:
There are basic NB-IoT examples for the UE in receive only mode. The full eNB is not available.