srsran/srsRAN_4G #99: error with type srslte_tdec_t

cedric-roux, 2017-09-11:
Hi,

I'm using the srsLTE libraries and I had an issue at runtime because I didn't use `-DLV_HAVE_SSE` in my makefile, so the type `srslte_tdec_t` was different from the one really compiled in srsLTE.

You should generate a kind of `config.h` containing the `#define` that were used at compilation time of srsLTE so that users of the library don't have to know that they have to put some of them manually in their project.

This, or maybe use the `pkg-config` logic and provide a `srslte.pc` file which has them inside. Or another automatic mechanism...

What do you think?

Regards,
Cédric.

ismagom, 2017-09-11:
Hi Cedric, I'm not sure I understood your point. If you are trying to disable SSE you should pass -DENABLE_SSE=OFF to CMake instead of LV_HAVE_SSE. The variable LV_HAVE_SSE is generated automatically by CMake if the target processor is compatible with SSE. 

Same would apply to ENABLE_AVX, ENABLE_AVX2 etc 

Is that what you are trying to achieve?

Regards
Ismael

cedric-roux, 2017-09-12:
Hi Ismael,

no I am using srsLTE as a library in another program. So I do:
```
#include <srslite/srslte.h>
```

Then I call the functions I need (like `srslte_pbch_init`, `srslte_ofdm_rx_slot`, etc.).

I compile my program like:
```
gcc -o prog prog.c -lsrslte_common -lsrslte_phy -lsrslte_upper -lsrslte_asn1 -lm -lfftw3f
```

And that sould be it.

The thing is, as a user of the library, I have no idea if it has been compiled with or without
`-DLV_HAVE_SSE`. Since this is known at the compilation time of srsLTE, I think this information
should be saved in some config.h file that is included at the top of srslte.h (ie. the user of the
library has nothing to do) and that defines or not all the constants like this one.

To be very precise, the problem is with the structure `srslte_tdec_t` defined in the file
`lib/include/srslte/phy/fec/turbodecoder.h`, we see:

```C
typedef struct SRSLTE_API {
#ifdef LV_HAVE_SSE
  srslte_tdec_sse_t tdec_sse;
#else
  float *input_conv; 
  srslte_tdec_gen_t tdec_gen;
#endif  
} srslte_tdec_t;
```

See, there is an `#ifdef` in there. This is the problem, because when I install the library,
the information that this `LV_HAVE_SSE` has been set or not at compilation time does
not exist at all. I have to know that the compilation logic did set it or not. This is not what
an user of a library expects.

I hope I'm clearer.

ismagom, 2017-09-12:
I see your point now. You are completely right. I think the best should be that any struct depends on LV_HAVE_SSE at all. This is something of the implementation and should be hidden to the user of the library. We'll spot those cases and remove them in the next release. 

Thanks a lot for your feedback.

closed 2017-10-06 by andrepuschmann