srsran/srsRAN_4G #111: Activity Monitor check gives negative time

ismagom, 2017-10-30:
Issue identified in the list: http://www.softwareradiosystems.com/pipermail/srslte-users/2017-October/001192.html

While the UE is doing some activity, RRC in srsENB computes last activity in function srsenb::rrc::ue::is_timeout() in:  https://github.com/srsLTE/srsLTE/blob/92f1ef19bc7cd50d3d4353e514274f043f482ab7/srsenb/src/upper/rrc.cc#L732 

resulting in negative value and the release of the UE: 

`08:53:53.563810 [RRC ] Warning User rnti=0x46 expired Activity deadline: -1:-561047891>3600:0 us`
`08:53:53.563835 [RRC ] Info    User rnti=0x46 timed out. Exists in s1ap=yes`
`08:53:53.563847 [S1AP] Info    User inactivity - RNTI:0x46`
`08:53:53.563870 [S1AP] Info    Sending UEContextReleaseRequest for RNTI:0x46`

nicolabui, 2017-10-30:
It only noticed it once. When it happened the c-RNTI got released

ismagom, 2017-10-30:
Can you attach enb.log?

nicolabui, 2017-10-30:
Attached the log with the negative activity deadline
[enb.log_negative.tar.gz](https://github.com/srsLTE/srsLTE/files/1428321/enb.log_negative.tar.gz)

ismagom, 2017-10-30:
Can you pull latest commit and test again?

nicolabui, 2017-10-30:
Here is the log.
[enb.log_after_fix.tar.gz](https://github.com/srsLTE/srsLTE/files/1428364/enb.log_after_fix.tar.gz)
I still have a disconnection, but it is not caused anymore by the negative deadline

ismagom, 2017-10-30:
This is then a different issue. What happens here is that there are Lates and Overflows in the USRP (maybe some process in background or some hardware interruption). You can see in the log in time stamps: 14:41:58.637823 to 14:41:58.639834. 

This shouldn't happen but may happen in some hardware. The UE then should PRACH and attempt a reconnection. That succeeds always when we use the Amarisoft EPC. For some reason this is not happening with OAI EPC. I'll open a new issue.

closed 2017-10-30 by ismagom