srsran/srsRAN_4G #113: srsENB does not reestablish the connection when using OAI EPC

ismagom, 2017-10-30:
1) UE looses connection
2) UE does PRACH + ConnectionReestablishment
3) srsENB rejects (not implemented) 
4) UE does PRACH + ConnectionRequest
5) IP traffic is restored in Amarisoft EPC but not in OAI EPC

This is issue identified in the list message: http://www.softwareradiosystems.com/pipermail/srslte-users/2017-October/001170.html

```
14:41:59.463204 [PHY0] Info    [09500] PRACH: 0/1, preamble=41, offset=0.0 us, peak2avg=706.4, max_offset=30.0 us
14:41:59.463388 [MAC ] Info    [09501] SCHED: Added user rnti=0x48
14:41:59.463406 [MAC ] Info    [09501] SCHED: Set bearer config lc_id=0, direction=3
14:41:59.463462 [RRC ] Info    Added new user rnti=0x48
14:41:59.463475 [MAC ] Info    [09501] RACH:  tti=9501, preamble=41, offset=0, temp_crnti=0x48
14:41:59.464005 [MAC ] Info    [09506] SCHED: RAR, ra_id=0, rnti=0x48, rarnti_idx=2, start_rb=0, n_rb=3, rar_grant_rba=52, rar_grant_mcs=0
14:41:59.464043 [PHY0] Info    [09502] PDCCH: DL DCI Format1A rnti=0x2, cce_index=0, L=4, tti_tx=9506
14:41:59.464058 [PHY0] Info    [09502] PDSCH: rnti=0x2, l_crb= 3, rb_start=0, harq=0, tbs=7, mcs=0, rv=0, tti_tx=9506
             0000: 69 00 00 68 0c 00 48 
14:41:59.465781 [MAC ] Info    [09512] SCHED: RAR tx rnti=0x48, pid=0, dci=0,0, grant=(2,5), n_rtx=0, tbs=7, bsr=0 (0-7)
14:41:59.473980 [PHY0] Info    [09512] PUSCH: rnti=0x48, prb=(2,5), tbs=7, mcs=0, rv=0, snr=21.9 dB, n_iter=1, crc=OK, dec_time=  33 us
             0000: 00 40 10 00 00 00 58 
14:41:59.474021 [PHY0] Info    [09512] PHICH: rnti=0x48, hi=1, I_lowest=2, n_dmrs=0, tti_tx=9516
14:41:59.474024 [PDCP] Info    RX SRB0 PDU
             0000: 40 10 00 00 00 58 
14:41:59.476610 [RRC ] Info    Removed user rnti=0x47
14:41:59.476629 [RRC ] Info    Rx SRB0 PDU
             0000: 40 10 00 00 00 58 
14:41:59.476651 [RRC ] Info    SRB0 - Rx: RRC Connection Request
             0000: 40 10 00 00 00 58 
14:41:59.476667 [RRC ] Info    Allocated SR resources for time-frequency slot (0, 0), N_pucch_sr=0, I_sr=16
14:41:59.476678 [RRC ] Info    Allocated CQI resources for time-frequency slot (0, 0), n_pucch_2=0, pmi_cfg_idx=37
14:41:59.476688 [MAC ] Info    [09522] Registering rnti=0x48 to PHY...
14:41:59.483686 [MAC ] Info    [09529] Done registering rnti=0x48 to PHY...
14:41:59.483708 [MAC ] Info    [09529] SCHED: Added user rnti=0x48
14:41:59.483720 [MAC ] Info    [09529] SCHED: Set bearer config lc_id=0, direction=3
14:41:59.483731 [MAC ] Info    [09529] SCHED: Set bearer config lc_id=1, direction=3
14:41:59.483757 [RLC ] Info    Adding radio bearer SRB1 with mode AM
14:41:59.483773 [RLC ] Info    SRB1 configured: t_poll_retx=45, poll_pdu=-1, poll_byte=-1000, max_retx_thresh=4, t_reordering=35, t_status_prohibit=0
14:41:59.483784 [PDCP] Info    Added bearer SRB1
14:41:59.483805 [RRC ] Info    SRB0 - rnti=0x48, Sending: RRC Connection Setup
             0000: 60 12 9b 2e 6a 1e 82 b2 65 2d 21 83 4c 00 02 56 
             0010: 00 04 20 
14:41:59.483821 [PDCP] Info    TX SRB0 SDU, do_security = false
             0000: 60 12 9b 2e 6a 1e 82 b2 65 2d 21 83 4c 00 02 56 
             0010: 00 04 20 
14:41:59.483835 [RLC ] Info    SRB0 Tx SDU
             0000: 60 12 9b 2e 6a 1e 82 b2 65 2d 21 83 4c 00 02 56 
             0010: 00 04 20 
14:41:59.483850 [RLC ] Info    Buffer state: rnti=0x48, lcid=0, tx_queue=19
14:41:59.483890 [MAC ] Info    [09526] SCHED: Added MAC Contention Resolution CE for rnti=0x48
14:41:59.483919 [MAC ] Info    [09526] SCHED: DL tx rnti=0x48, pid=0, mask=0x1fc0, dci=3,8, n_rtx=0, tbs=47, buffer=0
14:41:59.483932 [MAC ] Info    [09526] CE:    Added Contention Resolution ID=0x401000000058
14:41:59.483944 [RLC ] Info    SRB0 Complete SDU scheduled for tx. Stack latency: 0 us
14:41:59.483955 [RLC ] Info    TX SRB0, Transparent Mode PDU
             0000: 60 12 9b 2e 6a 1e 82 b2 65 2d 21 83 4c 00 02 56 
             0010: 00 04 20 
14:41:59.483992 [PHY0] Info    [09522] PDCCH: DL DCI Format1  rnti=0x48, cce_index=8, L=8, tti_tx=9526
14:41:59.484005 [PHY0] Info    [09522] PDSCH: rnti=0x48, l_crb=14, mask=0x1fc0, harq=0, tbs=47, mcs=0, rv=0, tti_tx=9526
             0000: 3c 20 13 1f 40 10 00 00 00 58 60 12 9b 2e 6a 1e 
             0010: 82 b2 65 2d 21 83 4c 00 02 56 00 04 20 00 00 00 
14:41:59.492080 [PHY0] Info    [09530] PUCCH: rnti=0x48, corr=0.99, n_pucch=20, n_prb=1, ack=1
14:41:59.503066 [PHY1] Info    [09541] PUCCH: rnti=0x48, corr=0.99, n_pucch=0, n_prb=23, sr=yes
14:41:59.503110 [MAC ] Info    [09549] SCHED: UL tx rnti=0x48, pid=5, dci=3,8, grant=(0,25), n_rtx=0, tbs=85, bsr=0 (512-85)
14:41:59.503149 [PHY1] Info    [09541] PDCCH: UL DCI Format0  rnti=0x48, cce_index=8, L=8, tti_tx=9545
14:41:59.510989 [PHY1] Info    [09549] PUSCH: rnti=0x48, prb=(0,25), tbs=85, mcs=0, rv=0, snr=13.0 dB, n_iter=1, crc=OK, dec_time= 141 us
             0000: 3d 3a 21 0f 1f 00 22 a0 00 00 20 10 09 8e 05 86 
             0010: 92 a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
14:41:59.511037 [PHY1] Info    [09549] PHICH: rnti=0x48, hi=1, I_lowest=0, n_dmrs=0, tti_tx=9553
14:41:59.511038 [RLC ] Info    SRB1 Rx data PDU SN: 0
             0000: 00 20 10 09 8e 05 86 92 a0 00 00 00 00 
[... 184 lines cut ...]
14:42:01.482134 [PHY0] Info    [01280] PUCCH: rnti=0x48, corr=1.01, n_pucch=0, n_prb=24, cqi=15
14:42:01.482834 [PHY1] Info    [01281] PUCCH: rnti=0x48, corr=0.09, n_pucch=0, n_prb=23, sr=no
14:42:01.503063 [PHY1] Info    [01301] PUCCH: rnti=0x48, corr=0.12, n_pucch=0, n_prb=23, sr=no
14:42:01.515670 [S1AP] Info    Received S1AP PDU
             0000: 00 17 00 11 00 00 02 00 63 00 04 00 05 00 01 00 
             0010: 02 40 02 00 00 
14:42:01.515710 [S1AP] Info    Received UEContextReleaseCommand
14:42:01.515726 [RRC ] Info    Releasing E-RABs for 0x46
14:42:01.515736 [RRC ] Warning Unrecognised rnti: 0x46
14:42:01.515752 [S1AP] Info    Sending UEContextReleaseComplete for RNTI:0x46
             0000: 20 17 00 0f 00 00 02 00 00 40 02 00 05 00 08 40 
             0010: 02 00 01 
14:42:01.515827 [S1AP] Info    UE context for RNTI:0x46 released
14:42:01.515849 [RRC ] Info    Received Release Complete rnti=0x46
14:42:01.515859 [RRC ] Error   Received ReleaseComplete for unknown rnti=0x46
14:42:01.521879 [PHY0] Info    [01320] PUCCH: rnti=0x48, corr=1.01, n_pucch=0, n_prb=24, cqi=15
14:42:01.522941 [PHY1] Info    [01321] PUCCH: rnti=0x48, corr=0.06, n_pucch=0, n_prb=23, sr=no
14:42:01.542228 [GTPU] Error   Unrecognized RNTI for DL PDU: 0x46 - dropping packet
14:42:01.542811 [PHY1] Info    [01341] PUCCH: rnti=0x48, corr=0.04, n_pucch=0, n_prb=23, sr=no
14:42:01.561991 [PHY0] Info    [01360] PUCCH: rnti=0x48, corr=1.01, n_pucch=0, n_prb=24, cqi=15
14:42:01.563040 [PHY1] Info    [01361] PUCCH: rnti=0x48, corr=0.01, n_pucch=0, n_prb=23, sr=no
14:42:01.582927 [PHY1] Info    [01381] PUCCH: rnti=0x48, corr=-0.04, n_pucch=0, n_prb=23, sr=no
14:42:01.602109 [PHY0] Info    [01400] PUCCH: rnti=0x48, corr=1.01, n_pucch=0, n_prb=24, cqi=15
14:42:01.603143 [PHY1] Info    [01401] PUCCH: rnti=0x48, corr=0.05, n_pucch=0, n_prb=23, sr=no
```

nicolabui, 2017-10-31:
In the log attached the following sequence of events happens:
- the mobile phone connects with RNTI 0x46 (connection working)
- srsenb disconnects the RNTI after 10 seconds of inactivity
- I woke the mobile phone which makes srsenb assign a new RNTI 0x47
- the connection does not work (I tried a ping which timeouts)
- I toggled the airplane mode on and off in the phone which makes the phone dsiconnecting RNTI 0x47 and performing a new random access
- srsend assigns a new RNTI 0x48
- connectvity working again (ping successful)
[enb.log_reconnect.tar.gz](https://github.com/srsLTE/srsLTE/files/1432409/enb.log_reconnect.tar.gz)

kewinrausch, 2018-01-29:
I tried this configuration with OAI EPC, develop branch.

To resume:
- Connect UE (RACH and everything done)
- Move UE out of range for 10 seconds
- Go back again in range of srsENB (**NO Airplane mode ON/OFF**)

Results are that the UE eventually (Nexus is slow in this process) reconnects again with eNB, and connection with EPC is re-established, but something withing the EPC went wrong, since I can see the MME but can't establish internet connection.

So UE reconnects again, but no Internet traffic.
I suppose this is a problem with OAI EPC, rather than with srsENB.

spencersevilla, 2018-04-05:
Hi Ismael and Kewin,

I'm also seeing the exact same pattern/bug with the OAI EPC (I'm using master branch) after idle disconnect/reconnect. I collected some logs and at a first glance, they look quite similar to yours. I share your suspicions that this is an OAI bug, because I've been able to recreate this pattern with two different eNBs (srsENB and a commercial product) as well as three different commercial UEs. I was wondering if either of you got any further on this, or figured out what was going on? Any hints or other ideas you could suggest? Just figured I'd check if there was any help you provide. Thank you!!!

spencersevilla, 2018-04-08:
Just in case anyone else finds this and/or is interested, I figured out the problem in the OAI EPC, and have a fix. When the UE goes idle, the MME tells the SPGW to release the context (relevant function is sgw_handle_release_access_bearers_request in sgw_handlers.c) but the SPGW forgets to tear down the old GTP tunnel. This means that when the UE wakes up, the MME tells the SPGW to create a new GTP tunnel, creation fails (because there already exists a tunnel), but the old tunnel has the wrong RNTI value. You can fix this by adding the line
`gtp_mod_kernel_tunnel_del(eps_bearer_entry_p->s_gw_teid_S1u_S12_S4_up, eps_bearer_entry_p->enb_teid_S1u);`
to the function sgw_release_all_enb_related_information in sgw_handlers.c right above the memset line. I am trying to push this to openair-cn soon, but hope this helps in the meantime!

ismagom, 2018-04-08:
Thanks for the feedback. Closing since it's not an srsUE bug

closed 2018-04-08 by ismagom