srsran/srsRAN_4G #116: Can't see ENB on my phone

tanatosos, 2017-11-01:
Hello,

I've managed to run srsenb on Ubuntu 14.04 with bladeRF.
This is what I have in `enb.conf`.
```
[enb]
enb_id = 0x19B
cell_id = 0x01
phy_cell_id = 1
tac = 0x0007
mcc = 001
mnc = 01
mme_addr = 192.168.1.100
gtp_bind_addr = 192.168.1.1
n_prb = 25

dl_earfcn = 6350
tx_gain = 50
rx_gain = 50

```

When I run it I'm getting this:

```
Reading configuration file enb.conf...
Opening USRP with args: 
Error opening UHD: code 11
Opening bladeRF...
Setting frequency: DL=811.0 Mhz, UL=852.0 MHz
set TX frequency to 811000000
set RX frequency to 852000000
Setting Sampling frequency 5.76 MHz
Set RX sampling rate 5.76 Mhz, filter BW: 5.00 Mhz

22:04:21.636807 [PHY0] Debug   [05841] Worker 0 running
22:04:21.636818 [PHY0] Debug   [05841] Processing pending_tti=5841
22:04:21.636872 [MAC ] Debug   [05845] SCHED: Allocated DCI L=2, ncce=0
22:04:21.636885 [MAC ] Debug   [05845] SCHED: SIB1, start_rb=0, n_rb=4, rv=0, len=15, period=8
22:04:21.636960 [PHY0] Debug   [05841] Sending to radio
22:04:21.637650 [PHY0] Debug   [05841] Settting TTI=5842, tx_mutex=6, tx_time=46:0.810002 to worker 1
22:04:21.637677 [PHY1] Debug   [05842] Worker 1 running
22:04:21.637751 [PHY1] Debug   [05842] Sending to radio
22:04:21.638520 [PHY0] Debug   [05841] Settting TTI=5843, tx_mutex=7, tx_time=46:0.811002 to worker 0
```

but I can't see it on my phone network scan (used Huawei P9 and OnePlus 3). Is this due 800MHz band?

I've also used commercial MCC and MNC but still no luck.
Did I misunderstand something?

tanatosos, 2017-11-02:
I've managed to get Test PLMN 4G to appear. I've used `dl_earfcn = 25` but it's pretty much unstable. Network keeps disappearing after a while and I guess this is because of bladeRF.

ismagom, 2017-11-04:
When you use the bladerf you need to calibrate the RF in the corresponding section of the enb.conf.

For best performance we recommend B2x0 usrps

closed 2017-11-04 by ismagom