srsran/srsRAN_4G #132: intrusive_ptr.hpp assertion fails

tblake84, 2018-01-08:
When trying to run the ENB, It fails to initialize and I receive the following error:

`Setting frequency: DL=739.0 Mhz, UL=709.0 MHz
Setting Sampling frequency 5.76 MHz

==== eNodeB started ===
Type <t> to view trace
srsenb: /usr/include/boost/smart_ptr/intrusive_ptr.hpp:174: T* boost::intrusive_ptr<T>::operator->() const [with T = uhd::transport::managed_send_buffer]: Assertion `px != 0' failed.
Aborted (core dumped)`

I am running this on the metal using Ubuntu Server 16.04.

ismagom, 2018-01-20:
This should be fixed in the last release. If you are running srsENB in an Intel Skylake Core, try disabling hyperthreading.

closed 2018-01-20 by ismagom

maiconkist, 2018-05-08:
Hi, 

this problem is happening to me with the last version in master (hash: 6b9f35ab6749c77545c8fbb46b3d9008d67394ca)

Log message is:
==== eNodeB started ===
Type <t> to view trace
srsenb: /usr/include/boost/smart_ptr/intrusive_ptr.hpp:174: T* boost::intrusive_ptr<T>::operator->() const [with T = uhd::transport::managed_send_buffer]: Assertion `px != 0' failed.

Ubuntu with Kernel: 4.4.0-45-generic #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux