srsran/srsRAN_4G #134: SIB configuration

mrlnc, 2018-01-17:
Hi all,

I think there are some problems with SIB configuration. Possible fix included.

## Default Config
This is from the SIB1 config in `sib.conf`:
```c
sched_info =
    (
        {
            si_periodicity = 16;
	    si_mapping_info = []; // comma-separated array of SIB-indexes (from 3 to 13). 
				   // Leave empty or commented to just scheduler sib2
        }
);
```
### Expected Behaviour
* only SIB1, SIB2 is transmitted
```
65535 LTE RRC DL_SCH SystemInformationBlockType1
65535 LTE RRC DL_SCH SystemInformation [ SIB2 ]
```

### Actual Behaviour
Despite no optional SIB was configured, srsenb transmits an empty SIB3, which is not announced in SIB1 `schedulingInfoList`.

```
65535 LTE RRC DL_SCH SystemInformationBlockType1
65535 LTE RRC DL_SCH SystemInformation [ SIB2 SIB3 ]
```

### Custom Config
If I attempt to transmit another SIB (assuming SIB4 exists) and use this config:
```c
sched_info =
    (
        {
            si_periodicity = 16;
	    si_mapping_info = [4]; // comma-separated array of SIB-indexes (from 3 to 13). 
				   // Leave empty or commented to just scheduler sib2
        }
);
```
then SIB4 is advertised in SIB1, but still only SIB2 and SIB3 are transmitted.

#### Expected Behaviour
```
65535 LTE RRC DL_SCH SystemInformationBlockType1
65535 LTE RRC DL_SCH SystemInformation [ SIB2 SIB4 ]
```

#### Actual Behaviour
```
65535 LTE RRC DL_SCH SystemInformationBlockType1
65535 LTE RRC DL_SCH SystemInformation [ SIB2 SIB3 ]
```

## Possible Solution
After some debugging, I think there is a off-by-one, that leads to SIB3 being included and configured SIBs ignored.
https://github.com/srsLTE/srsLTE/blob/dafcefb487c1453b1f0bd6e581c1c7f95dd92e79/srsenb/src/upper/rrc.cc#L144
should read
```cpp
      memcpy(&msg[i].sibs[j+j_start], &cfg.sibs[(int) sched_info[i-1].sib_mapping_info[j].sib_type+2], sizeof(LIBLTE_RRC_SYS_INFO_BLOCK_TYPE_STRUCT));
```

`i` and `j` were not particularly helpful, so I changed some variable names. See 97ef33ca9a8e01cc4d43b3a6863ca2261129b45a for a patch. I've not created a pull request, since I have not yet fully understood SIB scheduling, and cannot tell whether things work as intended.

mrlnc, 2018-01-18:
Is this comment still accurate? Multiple SIBs in one SI message seems to work for me
https://github.com/srsLTE/srsLTE/blob/4bbcfaa1dcbe347e4c99539bd1d3ec7a92176b4e/srsenb/src/mac/scheduler.cc#L396

ismagom, 2018-01-20:
Thanks for spotting this. Can you please create a pull request and I'll merge it? 

Can you also remove that comment in the pull request please? Thanks

closed 2018-01-20 by ismagom