srsran/srsRAN_4G #142: bladeRF calibration not implemented correctly

ralftoringo, 2018-02-02:
The current code calls the BladeRF calibration function bladerf_set_correction with 0 for each TX parameter regardless of what is set in the configuration of srsUE or srsENB. I was able to fix this by changing the data type from float to int16_t everywhere (which is what bladerf_set_correction expects).

Furthermore, despite being mentioned in the config files the rx calibration implementation is not finished. It is missing the lines that make it a valid config option in main.cc of both srsUE and srsENB.

However even with these fixes and calibrating the two bladeRFs correctly I am only able to establish a connection between two bladeRFs on very few occasions. SrsUE can find the srsENB signal, but srsENB does not receive anything from srsUE, despite a third SDR showing that data is being sent. The bladeRFs are connected directly to each other with attenuators and they use the same clock.

Connecting srsUE to a commercial base station using bladeRF works very reliably, but connecting a mobile phone to srsENB is also very unreliable.

1. Which SDR do you mainly use for srsENB?
2. It says in the readme that you tested srsLTE with bladeRF. Did you manage to get two of them to connect reliably to each other or use one reliably as base station? If so, do you have further recommendations on how to achieve that?

smarek, 2018-04-01:
I can confirm, do you have @ralftoringo your attempted patches anywhere public?

@ismagom it's been 2 months since creating this issue, could you please comment at least?

andrepuschmann, 2018-04-03:
Hey, unfortunately, our bladeRF support is somewhat rusty at this stage, simply because we don't have enough units and hence don't use them for development or testing. If others are willing to take over this job, please go ahead and submit config files, patches, etc. That's why it's open-source.

Kapulara, 2018-05-16:
We have a couple of units here. Might try to fix this.

aricwang88, 2018-05-30:
@Kapulara , I'm struggling and suffering from the bladerf calibration issue, both DC and IQ imbalance. I do as the intructions on the bladeRF's wiki, no result.  How to calibrate bladeRF, make it work for srsue? Thanks advance!!!!

aricwang88, 2018-05-30:
@andrepuschmann , pls gave us a hand.

aricwang88, 2018-06-03:
The root cause is: start burst tx twice, it cause the internal logic mess, I can rise a pull request to fix this if it possile.  How to get it work: 
1, Comment all the code in xxx_rx_dc_cal(), xxx_tx_dc_cal() as other SDRs.
2, add a boolean variable to indicate that the burst tx is started, for example   is_gBurstStarted=false;
When is_start_of_burst == true and is_gBurstStarted == false, then send burst tx and is_gBurstStarted = true.  When is_end_of_sburst == true, set is_gBurstStarted = false.
3, Recompile the code and run srsrenb once,  then stop and use bladerf-cli to calibrate it.
bladerf_cli -i
cal lms
cal dc rx
cal dc tx
exit
4, Run srsenb again, it should be normal.
Thanks.

andrepuschmann, 2018-07-03:
Hey @aricwang88 , please feel free to submit a PR for the issue. We'll then see if we can include it.

closed 2018-07-03 by andrepuschmann