srsran/srsRAN_4G #150: Failed to find UHD device

tblake84, 2018-02-15:
I am running srsenb in a docker container for testing.  Previous versions had other problems but this latest commit exhibits an inability to initiate the RF frontend.

```
root@30f6b07b726a:/opt/lte# srsenb enb.conf
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.009.007-0-g50839059

Warning: Failed to create thread with real-time priority. Creating it with normal priority: No such file or directory
Warning: Failed to create thread with real-time priority. Creating it with normal priority: No such file or directory
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...
Opening USRP with args: addr=10.10.210.141,master_clock_rate=30.72e6
-- X300 initialization sequence...
-- Determining maximum frame size... 1472 bytes.
-- Setup basic communication...
-- Loading values from EEPROM...
-- Setup RF frontend clocking...
Error opening UHD: code 44
Error opening RF device
Failed to find device UHD with args addr=10.10.210.141
```

Here is the relevant portion of the enb.conf
```
[rf]
dl_earfcn = 5780
tx_gain = 70
rx_gain = 40
device_name = UHD
device_args = addr=10.10.210.141
time_adv_nsamples = auto
burst_preamble_us = auto
```

andrepuschmann, 2018-02-16:
Just to double check. The problem also exists if you run srsENB from outside of the Docker container, right?

tblake84, 2018-04-20:
Sorry for the late response.  Yes, it occurs outside the docker container as well.  I have tried various versions of UHD (3.9.7, 3.9.10 and the latest 3.11.0.1) and all of them show the same error:
```
root@srslte:/opt/srslte# srsenb enb.conf
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.HEAD-0-ga1b5c4ae
Opening USRP with args: addr=10.10.210.147,master_clock_rate=30.72e6
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Determining maximum frame size...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Setup basic communication...
[INFO] [X300] Loading values from EEPROM...
[INFO] [X300] Setup RF frontend clocking...
Error opening UHD: code 44
Error opening RF device
Failed to find device UHD with args addr=10.10.210.147
```

andrepuschmann, 2018-04-23:
Thanks for checking again. We are working on it.

andrepuschmann, 2018-06-25:
Fixed in release 18.06

closed 2018-06-25 by andrepuschmann

hettipeti, 2018-07-05:
For People which use maybe still an old Version and get this error for the srsUE with "Error opening UHD: code 44":
For me the problem was, that I had an old FPGA Image for the USRP.
After downloading new images via uhd_images_downloader tool and uploading the new images to the USRP the problem was fixed.

pgoyal1, 2019-03-19:
Fixed in release 18.06 of what ? 
Also i tried installing latest USRP images but error still coming for me.

wilmar-arcila, 2020-02-05:
I'm using version 19.12.0 of srsLTE and getting the same error.
My device is USRP-2901 and using UHD 3.14.1.1-release

[image]

hanif81, 2020-03-04:
Hi, 
I am running srsenb on 18.04 ubuntu machine, but when i try to run it, it gives the the following error:
[image]
Could anyone please help me in this regard

twei7, 2020-03-05:
Having the exact same issue. However, deleting the build directory, recompiling and make install the srsLTE again solve the issue for me.

aes5001, 2020-03-24:
Hi guys, 
The same problem happened to me.
I did what "twei7" suggested, but did not work. Any idea what would be the problem?
I deploy srs LTE on ubuntu 18.04 cloud image, do you think this may cause the problem?

sadiqzq, 2021-10-13:
I am having the same issue when I run 

"osboxes@osboxes:~$ sudo srsenb
[sudo] password for osboxes:
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=default with args=default
[zmq] Error: RF device args are required for ZMQ no-RF module
/home/osboxes/srsRAN/lib/src/phy/rf/rf_imp.c.129: No compatible RF frontend found
Error initializing radio."

Can anybody guide me on this?