srsran/srsRAN_4G #152: PCAP corruption

mrlnc, 2018-02-22:
I frequently see corrupted PCAPs on `master`. Wireshark will either complain about the damaged PCAP file or show a very large, last packet.

The following is a workaround I apply on all my branches. I experienced not a single corrupted file over thousands of runs. Not sure about the performance implications.

```cpp
diff --git a/lib/include/srslte/common/pcap.h b/lib/include/srslte/common/pcap.h
index ce4fcf54..08151510 100644
--- a/lib/include/srslte/common/pcap.h
+++ b/lib/include/srslte/common/pcap.h
@@ -129,6 +129,7 @@ inline FILE *LTE_PCAP_Open(uint32_t DLT, const char *fileName)
 
     /* Write the file header */
     fwrite(&file_header, sizeof(pcap_hdr_t), 1, fd);
+    fflush(fd);
 
     return fd;
 }
@@ -208,6 +209,7 @@ inline int LTE_PCAP_MAC_WritePDU(FILE *fd, MAC_Context_Info_t *context,
     fwrite(&packet_header, sizeof(pcaprec_hdr_t), 1, fd);
     fwrite(context_header, 1, offset, fd);
     fwrite(PDU, 1, length, fd);
+    fflush(fd);
 
     return 1;
 }
@@ -243,6 +245,7 @@ inline int LTE_PCAP_NAS_WritePDU(FILE *fd, NAS_Context_Info_t *context,
     /* Now write everything to the file                            */
     fwrite(&packet_header, sizeof(pcaprec_hdr_t), 1, fd);
     fwrite(PDU, 1, length, fd);
+    fflush(fd);
 
     return 1;
 }
```

ismagom, 2018-02-22:
Have you seen any performance penalty due to flushing the disk every call? Can you create a pull request please?

mrlnc, 2018-02-22:
Actually I have not tested for performance. But this is on a powerful i7 and PCAPs are written to a ramdisk. Paging was not a problem, didn't test for high-speed DL-SCH traffic, though.

That's why I didn't create a pull request in the first place.

ismagom, 2018-02-22:
can you create it regardless? We'll test it and decide if to merge. Thanks!

laf0rge, 2018-02-22:
Why not simply flush on exit? Flushing every frame  can cause significant stalling depending on how busy your I/O subsystem is.
-- 
Sent from a mobile device. Please excuse my brevity.

mrlnc, 2018-02-22:
In my use case, I was running srsUE hundreds of times, automatically ending it. At the moment, exit seems pretty unstable to me, sometimes resulting in segfaults. That sometimes seemes to introduce errors to the PCAPs.

I don't propose to have this in `master`, but just want to bring this up in case others see this, too.

If srsUE fails when data is handed to `fwrite`, but without `fd` being flushed, is the data written to the file?

andrepuschmann, 2018-02-22:
No, it's not guaranteed. That's why you may end up with a corrupted file. The solution to this is to eliminate the bug that causes the segfault in the first place. IMHO even the flush on exit shouldn't be needed if the file handle is properly closed.

closed 2018-02-22 by andrepuschmann