srsran/srsRAN_4G #154: srsEPC: Attach

mrlnc, 2018-02-28:
Hi all,
my commercial UEs[1] I couldn't attach to srsENB + srsEPC. I think these are the issues:

### Attach Accept Result
Currently srsEPC sends `attach accept` with `LIBLTE_MME_EPS_ATTACH_RESULT_EPS_ONLY`. UEs will keep sending `Tracking Area Update` with update type `Combined TA/LA updating with IMSI attach` to request non-EPS services.

Changing `attach accept` result to `LIBLTE_MME_EPS_ATTACH_RESULT_COMBINED_EPS_IMSI_ATTACH` makes the UE proceed.

Fix in d21c6277f62610c2eefe773bfe5e07015511647b

### Attach Accept Message Corruption

`act_def_eps_bearer_context_req.connectivity_type_present` in the `attach accept` message is undefined, therefore the message is corrupted. One UE tolerates that, others are picky.

Fix in 8400ec645c78c0495dd1112d94934789a9bfb4c2

### Missing Service Request Handling

After Radio Link Failure or RRC inactivity timeout, UEs try to re-activate the NAS context and send a `Service Request`. srsEPC does currently not handle srsEPC and UE is stuck in a loop continuously sending Service Requests.

I 'implemented' (aka. copy & paste) Service Reject with `LIBLTE_MME_EMM_CAUSE_IMPLICITLY_DETACHED` to force the UE to re-attach. This leaves some resources at ENB and EPC, but should work as a temporary fix. Of course, re-establishment is not very smooth that way. I think I made some mistakes in the S1AP message handling, especially I don't understand what `MME_UE_S1AP_ID` to set at this stage of communication.

ccac32a9e5b6a15804e7bf29a6c03d4b0f0ee4ae and a6be68c9ebecaa59b1d73f4bb95fe09c8d7ca646

[1] Nexus 5, Huawei P9 lite, Honor 8

andrepuschmann, 2018-04-03:
I believe this is fixed now, right?

mrlnc, 2018-04-03:
Didn't have a chance to test yet. From what I've seen in the commits, it should be fixed. I'll close the issue and reopen if necessary.

closed 2018-04-03 by mrlnc

alvasMan, 2018-04-03:
The first 3 issues in this should be fixed. I make some comments on each individual issue below:

**Attach Accept Result**
srsEPC now echoes the attach type back to the UE. This will not break srsUE which expects an EPS only attach and lets commercial UEs perform a Combined Attach.

**Attach Accept Message Corruption**
Fixed, thanks for reporting that.

**Missing Service Request Handling**
Service request handling was added in version 18.3. This means that service request from the UE will re-establish the connection, without the user having to re-attach. I've tested this with a Moto G4 plus and it worked correctly.

**APN / Bearer Activation**
At the moment the APN must be set to match the APN at the EPC. I am able to replicate the issue you are having, but I haven't had much time look into what is the correct way of handling this. Please set a matching APN at the UE for now, and I will look into this.

@mrlnc, could you confirm if the latest version fixes the issues 1-3 for you? Particularly if the service requests are being handled correctly? If it does, we could close this issue and open another issue just for the APN.