srsran/srsRAN_4G #156: Run UE and ENB,after a while,around 1~3 minutes,downlink is not reachable.

liuzhikui, 2018-03-07:
When I run UE and ENB in different Hosts,and transfer video from UE to ENB using VLC software,after 2~3 minutes,video transmission was stopped.And I used Ping command in ENB side,Ping was not reachable.I used  Wireshark to captrue network data packets in UE side and ENB side,found that packet doesn't reach UE.Then,I used Ping in UE sdie,the packet can reach ENB,but UE doesn't recieve the reply.

**Ping is not reachable!!**
[image]

**Wireshark show information:**
[image]

pgorczak, 2018-03-13:
I've been observing the same behavior. However the time until failure was anywhere between 10 seconds and 15 minutes.

To reiterate: silent failure (also a commercial UE still thinks it's attached) after which only the uplink keeps working.

sharanerukulla, 2018-03-14:
Even I've been observing the same behavior, video transmission comes to a halt in less than 1-2 minutes. UE goes to RRC IDLE state and ping doesn't work.

andrepuschmann, 2018-03-14:
Can you send a UE and/or eNB log along with the error? Also, does the same behavior also occur with other eNBs, e.g. Amarisoft? If so, it would be a UE issue rather than eNB. I know @pgorczak said it also happens with commercial UE but that could also be something else.

pgorczak, 2018-03-14:
I'll post an eNB log later. Should I also include the pcap?

andrepuschmann, 2018-03-14:
Preferably both, yes, UE and eNB in info (PHY and RRC if it doesn't get too big) and pcap. Thanks

pgorczak, 2018-03-14:
I haven't been able to reproduce it with the current master. Can @sharanerukulla or @liuzhikui chime in?

pgorczak, 2018-03-14:
Okay, I've encountered the bug after the system ran flawlessly for a couple of hours. Had to upload the logs to [Google Drive](https://drive.google.com/file/d/1MeO_yS2gs4P5gOamqwp8xVXUpSuTyaL3/view?usp=sharing).

In this instance, the connection failed quite early. I ran iperf after attaching, which worked for about 10 seconds. I could still see UE-originated ping requests and replies at the `srs_spgw_sgi` interface but the replies didn't make it back to the UE. The UE is a *Sierra Wireless MC7455* btw.

Happy to provide more info if you need it!

andrepuschmann, 2018-03-14:
Great, thanks. We'll investigate the issue and get back.

sharanerukulla, 2018-03-14:
I was able to reproduce it with current master, video streaming stops after a couple of minutes and ping also doesn't seem to work.
In my case `srsue` and `srsenb` were running on two different USRP B210's.
I was using VLC to stream data on a TCP port on ENB and was playing the same on the UE.
Attaching UE, ENB, EPC logs, pcap files and console messages.
[srsLTE_Logs.zip](https://github.com/srsLTE/srsLTE/files/1813478/srsLTE_Logs.zip)

Do let me know incase you need any additional information.

liuzhikui, 2018-03-15:
Do you observe that "throw length_error" happen?

pgorczak, 2018-03-15:
One more thing I've observed: During reliable runs, I got very few *Underflow* and *Late* warnings from srsENB with a consistently higher round trip time (~60 ms). In error prone runs, there are many warnings with lower RTTs around 30 ms.

pgorczak, 2018-03-15:
I'll give that a shot and report back tomorrow :thumbsup:

pgorczak, 2018-03-17:
Alright I made some progress! Haven't encountered the error during several hours of testing.

Here are some suggestions for testing:

* follow the frequency scaling instructions above
* set your CPU power management to performance (BIOS)
* use a low latency kernel
* run iperf3 (TCP) and make sure
    * you don't get too many retransmissions (I got some every couple of seconds which I blame on congestion control)
    * your final throughput is close to the [TBS](https://www.rohde-schwarz.com/rsLTEResourceAllocation/index.php?toolID=TransportBlockSize_DL) * 1000 [MBit/s] (I got 90%)

There was a single short hiccup within an hour long test that caused a service request but didn't affect the TCP connection.

closed 2018-05-01 by ismagom

GeorgeJi-ca, 2021-11-07:
Hi liuzhikui and All, 
I’m running srsLTE End-to-end in different PCs, and transferring video from UE to ENB using VLC.  The running status is looking well (See attached pictures).  In UE side, by running: $ vlc -vvv  videofile.mp4 --sout tcp://172.16.0.1:8080 --loop, we can see the streaming data (brate=14M) on UL.  But in ENB side, by using: $ vlc -vvv mmsh://172.16.0.2:8080, VLC is unable to open and receive the video files always. 
Can you point the problems and give me the suggestions?
 Thanks!

ENB side:
[image]

UE side:
[image]