srsran/srsRAN_4G #157: terminate called after throwing an instance of 'std::length_error'   what():  basic_string::_S_create

nitha2000, 2018-03-16:
Hello,
this issue still exists in the latest load that I cloned a week ago. Also, it's reproducible and i always get this crash. It's sometimes accompanied with "double pointer corruption" message as well.

Clue: I'm using custom frequency and whenever the UL/DL frequency gap is higher, it crashes all the time. My Freq. are 400MHz DL and 1700MHz UL in the UE. enb is from amarisoft. I can provide more info if it's needed. Any ideas waht's going on....
thanks,

 
--- command='./srsue ./ue.conf' version=17.12.0 signal=11 date='15/03/2018 17:25:21' ---
	./srsue() [0x5a70ba]
	/lib/x86_64-linux-gnu/libc.so.6(+0x36cb0) [0x7fdba99b0cb0]
	/usr/local/lib/libuhd.so.003(uhd_tx_streamer_send+0x5e) [0x7fdba7c3ad2e]
	/home/user/Downloads/srsLTE_clone_Mar_08_2018/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(rf_uhd_send_timed_multi+0x194) [0x7fdbaa76e634]
	/home/user/Downloads/srsLTE_clone_Mar_08_2018/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(srslte_rf_send_timed_multi+0x2e) [0x7fdbaa76bcce]
	./srsue(_ZN6srslte5radio2txEPPvj18srslte_timestamp_t+0xd1) [0x613741]
	./srsue(_ZN6srslte5radio9tx_singleEPvj18srslte_timestamp_t+0x2e) [0x61380e]
	./srsue(_ZN5srsue11phch_common10worker_endEjbPCfj18srslte_timestamp_t+0xfb) [0x53530b]
	./srsue(_ZN5srsue11phch_worker8work_impEv+0x6bc) [0x54301c]
	./srsue(_ZN6srslte11thread_pool6worker10run_threadEv+0x31) [0x570431]
	./srsue(_ZN6thread21thread_function_entryEPv+0x9) [0x508bc9]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x8184) [0x7fdbab13d184]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d) [0x7fdba9a7437d]

nitha2000, 2018-04-12:
I tested the latest load and the issue still exists. The changes indicated in the report does not fix the issue.I can easily re-produce this issue.

 
 
Closed #157.

ismagom, 2018-04-13:
can you attach ue.conf, ue.log and exact command you use? I think in the previous issue you were not running as root. Can you run as root?

nitha2000, 2018-04-13:
Hi,thanks. Yes, I'll send you both.Yes, I was always running as root but the srsLTE_crash report does not include "sudo" in front of the actual cmd.

nitha2000, 2018-04-16:
Hi,
The latest load (cloned today, srsue with Amarisoft enb, 20MHz BW, iperf3 TP ~25Mbps) crashed after 45 minutes with the following error msg. but the ue.log is huge 10G.

Error on console....
------------------------
sudo ./srsue ./ue.conf
.
.
srsue: /usr/include/boost/smart_ptr/intrusive_ptr.hpp:162: T* boost::intrusive_ptr<T>::operator->() const [with T = uhd::transport::managed_send_buffer]: Assertion `px != 0' failed.
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...

srsLTE backtrace....
--------------------------

--- command='./srsue ./ue.conf' version=18.3.1 signal=6 date='16/04/2018 09:55:07' ---
        ./srsue() [0x597c5a]
        /lib/x86_64-linux-gnu/libc.so.6(+0x36cb0) [0x7f580275ccb0]
        /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x37) [0x7f580275cc37]
        /lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7f5802760028]
        /usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x155) [0x7f5802d61535]
        /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5e6d6) [0x7f5802d5f6d6]
        /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5e703) [0x7f5802d5f703]
        /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5e922) [0x7f5802d5f922]
        /usr/local/lib/libuhd.so.003(+0x70f59e) [0x7f5800d1b59e]
        /usr/local/lib/libuhd.so.003(+0x70f72b) [0x7f5800d1b72b]
        /usr/local/lib/libuhd.so.003(+0x70cc4b) [0x7f5800d18c4b]
        /usr/local/lib/libuhd.so.003(+0x70d2e6) [0x7f5800d192e6]
        /usr/local/lib/libuhd.so.003(+0x17966f) [0x7f580078566f]
        /usr/local/lib/libuhd.so.003(+0x49fa11) [0x7f5800aaba11]
        /usr/local/lib/libuhd.so.003(uhd_tx_streamer_send+0xdb) [0x7f58008714cc]
        /home/user/Downloads/srsLTE_clone_Apr_16_2018/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(rf_uhd_send_timed_multi+0x17f) [0x7f580351a46f]
        /home/user/Downloads/srsLTE_clone_Apr_16_2018/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(srslte_rf_send_timed_multi+0x2e) [0x7f5803517c0e]
        ./srsue(_ZN6srslte5radio2txEPPvj18srslte_timestamp_t+0xd1) [0x606421]
        ./srsue(_ZN5srsue5prach4sendEPN6srslte5radioEff18srslte_timestamp_t+0x1c1) [0x52f441]
        ./srsue(_ZN5srsue9phch_recv10run_threadEv+0x980) [0x533ea0]
        ./srsue(_ZN6thread21thread_function_entryEPv+0x9) [0x509579]
        /lib/x86_64-linux-gnu/libpthread.so.0(+0x8184) [0x7f5803ee9184]
        /lib/x86_64-linux-gnu/libc.so.6(clone+0x6d) [0x7f580282037d]

--- command='./srsue ./ue.conf' version=18.3.1 signal=6 date='16/04/2018 09:55:07' ---
        ./srsue() [0x597c5a]
        /lib/x86_64-linux-gnu/libc.so.6(+0x36cb0) [0x7f580275ccb0]
        /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x37) [0x7f580275cc37]
        /lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7f5802760028]
        /lib/x86_64-linux-gnu/libc.so.6(+0x2fbf6) [0x7f5802755bf6]
        /lib/x86_64-linux-gnu/libc.so.6(+0x2fca2) [0x7f5802755ca2]
        /usr/local/lib/libuhd.so.003(+0x158e0e) [0x7f5800764e0e]
        /usr/local/lib/libuhd.so.003(+0x70e987) [0x7f5800d1a987]
        /usr/local/lib/libuhd.so.003(+0x70eb48) [0x7f5800d1ab48]
        /usr/local/lib/libuhd.so.003(+0xde65d) [0x7f58006ea65d]
        /usr/local/lib/libuhd.so.003(+0x70c6d5) [0x7f5800d186d5]
        /usr/local/lib/libuhd.so.003(+0x70c761) [0x7f5800d18761]
        /usr/local/lib/libuhd.so.003(+0x70ccc9) [0x7f5800d18cc9]
        /usr/local/lib/libuhd.so.003(+0xde65d) [0x7f58006ea65d]
        /usr/local/lib/libuhd.so.003(+0x614ec9) [0x7f5800c20ec9]
        /usr/local/lib/libuhd.so.003(+0x615bff) [0x7f5800c21bff]
        /usr/local/lib/libuhd.so.003(+0x63aab1) [0x7f5800c46ab1]
        /usr/local/lib/libuhd.so.003(+0xde65d) [0x7f58006ea65d]
        /usr/local/lib/libuhd.so.003(+0x23b455) [0x7f5800847455]
        /usr/local/lib/libuhd.so.003(+0x23a567) [0x7f5800846567]
        /usr/local/lib/libuhd.so.003(+0x2841c7) [0x7f58008901c7]
        /usr/local/lib/libuhd.so.003(+0x284227) [0x7f5800890227]
        /lib/x86_64-linux-gnu/libc.so.6(+0x3c1a9) [0x7f58027621a9]
        /lib/x86_64-linux-gnu/libc.so.6(+0x3c1f5) [0x7f58027621f5]
        ./srsue() [0x597ff2]
        /lib/x86_64-linux-gnu/libc.so.6(+0x36cb0) [0x7f580275ccb0]
        /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x37) [0x7f580275cc37]
        /lib/x86_64-linux-gnu/libc.so.6(abort+0x148) [0x7f5802760028]
        /usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x155) [0x7f5802d61535]
        /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5e6d6) [0x7f5802d5f6d6]
        /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5e703) [0x7f5802d5f703]
        /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x5e922) [0x7f5802d5f922]
        /usr/local/lib/libuhd.so.003(+0x70f59e) [0x7f5800d1b59e]
        /usr/local/lib/libuhd.so.003(+0x70f72b) [0x7f5800d1b72b]
        /usr/local/lib/libuhd.so.003(+0x70cc4b) [0x7f5800d18c4b]
        /usr/local/lib/libuhd.so.003(+0x70d2e6) [0x7f5800d192e6]
        /usr/local/lib/libuhd.so.003(+0x17966f) [0x7f580078566f]
        /usr/local/lib/libuhd.so.003(+0x49fa11) [0x7f5800aaba11]
        /usr/local/lib/libuhd.so.003(uhd_tx_streamer_send+0xdb) [0x7f58008714cc]
        /home/user/Downloads/srsLTE_clone_Apr_16_2018/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(rf_uhd_send_timed_multi+0x17f) [0x7f580351a46f]
        /home/user/Downloads/srsLTE_clone_Apr_16_2018/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(srslte_rf_send_timed_multi+0x2e) [0x7f5803517c0e]
        ./srsue(_ZN6srslte5radio2txEPPvj18srslte_timestamp_t+0xd1) [0x606421]
        ./srsue(_ZN5srsue5prach4sendEPN6srslte5radioEff18srslte_timestamp_t+0x1c1) [0x52f441]
        ./srsue(_ZN5srsue9phch_recv10run_threadEv+0x980) [0x533ea0]
        ./srsue(_ZN6thread21thread_function_entryEPv+0x9) [0x509579]
        /lib/x86_64-linux-gnu/libpthread.so.0(+0x8184) [0x7f5803ee9184]
        /lib/x86_64-linux-gnu/libc.so.6(clone+0x6d) [0x7f580282037d]

ismagom, 2018-04-16:
Thanks for your debugging help. ue.log is not necessary don't worry. 

What USRP are you using?

What UHD version are you using? Can you try UHD 3.9LTS from this PPA: https://launchpad.net/~ettusresearch/+archive/ubuntu/uhd-3.9.lts

Thanks

nitha2000, 2018-04-16:
eNB side (N210) and UE side (B210). UHD version is; 3.010.003

nitha2000, 2018-04-18:
OK. I tried the UHD load you suggested, 3.9LTS. It also crashed after 20mins (20MHz BW), and the performance is not that great compare to the latest uhd version.
Please let me know if you want me to run any other tests to find out the culprit. thanks,

andrepuschmann, 2018-06-25:
This should be fixed as of release 18.06.

closed 2018-06-25 by andrepuschmann