srsran/srsRAN_4G #161: MAC Errors in 18.03 (but connection works)

pgorczak, 2018-04-03:
While running tests with the newest master srsENB, I'm getting pairs of the following error messages

    [MAC ] [E] [05249] Trying to write packet with invalid number of subheaders (nof_subheaders=0).
    [MAC ] [E] [05249] Error! PDU was not generated (rnti=0x0046, tb=0)

They appear every 2–10 seconds or so. Sometimes they're followed by another message

    [MAC ] [W] [05277] SCHED: discarting TB 0 pid=3, tti=5273, maximum number of retx exceeded (4)

Meanwhile the connection seems to work just fine.

pgorczak, 2018-04-10:
Another similar case: If the signal strength has been bad, the eNB sometimes keeps spamming the MAC warning message mentioned above forever

    [MAC ] [W] [05277] SCHED: discarting TB 0 pid=3, tti=5273, maximum number of retx exceeded (4)

closed 2018-05-01 by ismagom

aleksander0m, 2019-12-17:
@ismagom I wonder why this issue was closed without further comments?

ismagom, 2019-12-17:
We went through a large refactor and it didnt apply anymore. Sorry for not giving proper explanation. Reopen if you still see it and provide updated information please. Thanks

aleksander0m, 2019-12-17:
@ismagom not that I believe there is any error, but my log running the eNB is spamming these logs quite often:
```
14:29:34.744700 [MAC ] [W] [ 4393] SCHED: discarting TB 0 pid=1, tti=4393, maximum number of retx exceeded (4)
14:29:34.804731 [MAC ] [W] [ 4453] SCHED: discarting TB 0 pid=5, tti=4453, maximum number of retx exceeded (4)
14:29:34.864775 [MAC ] [W] [ 4513] SCHED: discarting TB 0 pid=1, tti=4513, maximum number of retx exceeded (4)
14:29:34.924623 [MAC ] [W] [ 4573] SCHED: discarting TB 0 pid=5, tti=4573, maximum number of retx exceeded (4)
14:29:34.984655 [MAC ] [W] [ 4633] SCHED: discarting TB 0 pid=1, tti=4633, maximum number of retx exceeded (4)
```
Just trying to get an idea of why those would happen.

ismagom, 2019-12-17:
That just means that there are too many retransmissions, probably due to bad channel conditions. To get further insights increase the phy log level to info. Then maybe use the list to ask about the possible cause.

pgorczak, 2019-12-17:
Can confirm that in newer versions I only get the warning when it's to be expected due to a bad channel.