srsran/srsRAN_4G #164: epc hangs and/or dumps core on termination

jgiovatto, 2018-04-04:
epc frequently hangs and/or dumps core on ctrl-c, noticed when running logs to stdout or file.
backtrace below

```
#3  __pthread_cond_signal (cond=cond@entry=0x7ffe5cf56dc0) at pthread_cond_signal.c:47
#4  0x000000000046a8c6 in srslte::logger_file::log (this=this@entry=0x7ffe5cf56d60, msg=msg@entry=0x12882f0) at /lib/src/common/logger_file.cc:78
#5  0x000000000046b210 in srslte::logger_file::~logger_file (this=0x7ffe5cf56d60, __in_chrg=<optimized out>) at lib/src/common/logger_file.cc:45
```
seems to be related to log in destructor possible race condition

workaround below seems to help

```diff --git a/lib/src/common/logger_file.cc b/lib/src/common/logger_file.cc
index a911ae20..dfc34805 100644
--- a/lib/src/common/logger_file.cc
+++ b/lib/src/common/logger_file.cc
@@ -43,11 +43,14 @@ logger_file::logger_file()
 
 logger_file::~logger_file() {
   not_done = false;
-  log(new std::string("Closing log\n"));
   if(inited) {
+    pthread_mutex_lock(&mutex);
+    pthread_cond_signal(&not_empty);
+    pthread_mutex_unlock(&mutex);
     wait_thread_finish();
     flush();
     if (logfile) {
+      fprintf(logfile, "Closing log\n");
       fclose(logfile);
     }
   }
@@ -84,6 +87,12 @@ void logger_file::run_thread() {
     pthread_mutex_lock(&mutex);
     while(buffer.empty()) {
       pthread_cond_wait(&not_empty, &mutex);
+
+      if(buffer.empty() && not_done == false)
+        {
+          printf("Log thread done\n");
+          return;
+        }
     }
     str_ptr s = buffer.front();
     pthread_cond_signal(&not_full);
```

jgiovatto, 2018-04-05:
Yes

andrepuschmann, 2018-07-03:
Closing issue, hopefully (difficulties reproducing it) solved in next release.

closed 2018-07-03 by andrepuschmann