srsran/srsRAN_4G #165: epc shuts down on SIGPIPE on Fedora 27

jgiovatto, 2018-04-04:
restarting ue and enb nodes eventually causes epc to throw SIGPIPE which is not handled.
start epc, enb, ue, wait for connection
kill ue, enb,
start enb, ue new session in epc catchs sigpipe which is terminal.

note signal seems to shows up on the next successful read in the new session so its not obvious to the caller.

```
8:08:52.432798 [S1AP] [D] Waiting for SCTP Msg
18:08:52.631697 [S1AP] [W] Previous connection broken from SCTP socket
18:08:52.631765 [S1AP] [I] Received S1AP msg. Size: 38
18:08:52.631794 [S1AP] [I] Received Succeseful Outcome PDU
18:08:52.631815 [S1AP] [I] Received Initial Context Setup Response.
```
gdb

```
Thread 2 "srsepc" received signal SIGPIPE, Broken pipe.
[Switching to Thread 0x7ffff5966700 (LWP 29037)]
0x00007ffff7bcbee4 in __libc_sendmsg (fd=3, msg=msg@entry=0x7ffff57144b0, flags=flags@entry=0) at ../sysdeps/unix/sysv/linux/sendmsg.c:28
28	  return SYSCALL_CANCEL (sendmsg, fd, msg, flags);
```
temp workaround below

```diff --git a/srsepc/src/main.cc b/srsepc/src/main.cc
index e2c00675..074cf008 100644
--- a/srsepc/src/main.cc
+++ b/srsepc/src/main.cc
@@ -275,6 +275,7 @@ main (int argc,char * argv[] )
   signal(SIGINT, sig_int_handler);
   signal(SIGTERM, sig_int_handler);
   signal(SIGKILL, sig_int_handler);
+  signal(SIGPIPE, SIG_IGN);

   all_args_t args;
   parse_args(&args, argc, argv); 
diff --git a/srsepc/src/mme/mme.cc b/srsepc/src/mme/mme.cc
index 1a385d8d..1c960b1a 100644
--- a/srsepc/src/mme/mme.cc
+++ b/srsepc/src/mme/mme.cc
@@ -135,6 +135,11 @@ mme::run_thread()
     m_s1ap_log->debug("Waiting for SCTP Msg\n");
     pdu->reset();
     rd_sz = sctp_recvmsg(s1mme, pdu->msg, sz,(struct sockaddr*) &enb_addr, &fromlen, &sri, &msg_flags);
+
+    if (errno == EPIPE){
+      m_s1ap_log->warning("Previous connection broken from SCTP socket\n");
+    }
+
     if (rd_sz == -1 && errno != EAGAIN){
       m_s1ap_log->error("Error reading from SCTP socket: %s", strerror(errno));
     }
```

andrepuschmann, 2018-04-05:
Thanks for spotting this. @alvasMan can we reproduce this with Ubuntu?

jgiovatto, 2018-04-05:
FYI, seems to only be an issue on the epc. Restarting the epc does not seem to effect the enb.

alvasMan, 2018-04-05:
No I can't seem to replicate this behavior in Ubuntu 16.04.
I'll investigate this next week on a Fedora machine. @jgiovatto does this *always* happens? Or just sometimes?

jgiovatto, 2018-04-05:
Happens every time.

eschreiber-alink, 2018-04-06:
I'm able to reproduce on Ubuntu 16.04 by running srsepc in gdb ...

# gdb sudo srsepc
(gdb) run epc.conf

Then same sequence as initially reported:
1. start enb, ue, wait for connection
2. kill ue, kill enb
3. start enb, start ue

Thread 3 "srsepc" received signal SIGPIPE, Broken pipe.
[Switching to Thread 0x7ffff6077700 (LWP 11847)]
0x00007ffff7bcaa6d in sendmsg () at ../sysdeps/unix/syscall-template.S:84

alvasMan, 2018-06-22:
Joe, are you sure that the steps to replicate this are not the following:

1 - Startup EPC, eNB and UE
2 - Kill eNB first, UE second.
3 - Restart eNB and UE?

There was a bug in the EPC where the EPC did not clean the UE context properly when the eNB was terminated while having users attached. If that is the issue, it will be fixed in the next release.

jgiovatto, 2018-06-22:
You may be correct, with the order above I do see SIGPIPE, when only enb ue interaction then no issues seen.

andrepuschmann, 2018-06-25:
Fixed in new release,

closed 2018-06-25 by andrepuschmann