srsran/srsRAN_4G #170: enb crash on setup failure

jgiovatto, 2018-04-17:
back trace below
```
 ./srsenb(_Z11liblte_packP21LIBLTE_BIT_MSG_STRUCTP22LIBLTE_BYTE_MSG_STRUCT+0x72) [0x54fae2]
        ./srsenb(_Z25liblte_s1ap_pack_s1ap_pduP27LIBLTE_S1AP_S1AP_PDU_STRUCTP22LIBLTE_BYTE_MSG_STRUCT+0xd5) [0x5a3e15]
        ./srsenb(_ZN6srsenb4s1ap31send_initial_ctxt_setup_failureEt+0x3c1) [0x4ac061]
        ./srsenb(_ZN6srsenb3rrc2ue34notify_s1ap_ue_ctxt_setup_completeEv+0xd5) [0x4a0a35]
        ./srsenb(_ZN6srsenb3rrc2ue13parse_ul_dcchEjPN6srslte13byte_buffer_tE+0x1e3) [0x4a31c3]
        ./srsenb(_ZN6srsenb3rrc13parse_ul_dcchEtjPN6srslte13byte_buffer_tE+0xab) [0x4a671b]
```
type mismatch in call to  liblte_s1ap_pack_s1ap_pdu proposed fix below ...

```
--- a/srsenb/src/upper/s1ap.cc
+++ b/srsenb/src/upper/s1ap.cc
@@ -986,7 +986,7 @@ bool s1ap::send_initial_ctxt_setup_failure(uint16_t rnti)
   fail->Cause.choice.radioNetwork.ext = false;
   fail->Cause.choice.radioNetwork.e   = LIBLTE_S1AP_CAUSERADIONETWORK_UNSPECIFIED;
 
-  liblte_s1ap_pack_s1ap_pdu(&tx_pdu, (LIBLTE_BYTE_MSG_STRUCT*)&buf);
+  liblte_s1ap_pack_s1ap_pdu(&tx_pdu, (LIBLTE_BYTE_MSG_STRUCT*)buf);
```

also moved log after sanity check in switch statement since call
 ```
toliblte_rrc_rlc_mode_text[cnfg.rlc_mode]
```
will fail badly if cnfg.rlc.mode is out of range due to current unknown reason.

```
 /lib64/libc.so.6(_IO_vfprintf+0x200b)
 /lib64/libc.so.6(vasprintf+0xb9)
 srsenb(_ZN6srslte10log_filter7warningEPKcz+0x9c)
 srsenb(_ZN6srslte3rlc10add_bearerEjNS_19srslte_rlc_config_tE+0xba)
 srsenb(_ZN6srsenb3rlc10add_bearerEtjN6srslte19srslte_rlc_config_tE+0x126)
 srsenb(_ZN6srsenb3rrc2ue22send_connection_reconfEPN6srslte13byte_buffer_tE+0x2da)
 srsenb(_ZN6srsenb3rrc2ue13parse_ul_dcchEjPN6srslte13byte_buffer_tE+0x1ee)
 srsenb/src/emu_srsenb(_ZN6srsenb3rrc13parse_ul_dcchEtjPN6srslte13byte_buffer_tE+0xab)
```

```--- a/lib/src/upper/rlc.cc
+++ b/lib/src/upper/rlc.cc
@@ -260,8 +260,6 @@ void rlc::add_bearer(uint32_t lcid, srslte_rlc_config_t cnfg)
   }
 
   if (!rlc_array[lcid].active()) {
-    rlc_log->warning("Adding radio bearer %s with mode %s\n",
-                  rrc->get_rb_name(lcid).c_str(), liblte_rrc_rlc_mode_text[cnfg.rlc_mode]);
     switch(cnfg.rlc_mode)
     {
     case LIBLTE_RRC_RLC_MODE_AM:
@@ -280,6 +278,8 @@ void rlc::add_bearer(uint32_t lcid, srslte_rlc_config_t cnfg)
       rlc_log->error("Cannot add RLC entity - invalid mode\n");
       return;
     }
+   rlc_log->warning("Adding radio bearer %s with mode %s\n",
+                  rrc->get_rb_name(lcid).c_str(), liblte_rrc_rlc_mode_text[cnfg.rlc_mode]);
   } else {
     rlc_log->warning("Bearer %s already created.\n", rrc->get_rb_name(lcid).c_str());
   }
```

jgiovatto, 2018-04-17:
noticed this error log where I added the timer id
```
Error releasing timer 0: nof_used_timers=0, nof_timers=128
```
``
--- a/lib/include/srslte/common/timers.h
+++ b/lib/include/srslte/common/timers.h
@@ -145,7 +145,7 @@ public:
       used_timers[i] = false;
       nof_used_timers--;
     } else {
-      fprintf(stderr, "Error releasing timer: nof_used_timers=%d, nof_timers=%d\n", nof_used_timers, nof_timers);
+      fprintf(stderr, "Error releasing timer %u: nof_used_timers=%d, nof_timers=%d\n", i, nof_used_timers, nof_timers);
     }
   }
   uint32_t get_unique_id() {
```
and this condition where the mode is not valid

```
--- a/lib/include/srslte/upper/rlc_interface.h
+++ b/lib/include/srslte/upper/rlc_interface.h
@@ -115,6 +115,7 @@ public:
         um.rx_mod              = (RLC_UMD_SN_SIZE_5_BITS == um.rx_sn_field_length) ? 32 : 1024;
         break;
       default:
+         fprintf(stderr, "%s %s RLC_MODE NOT VALID\n", __FILE__, __func__);
         // Handle default case
         break;
     }
```

andrepuschmann, 2018-06-25:
Fixed. Thanks.

closed 2018-06-25 by andrepuschmann