srsran/srsRAN_4G #201: Support XDG Base Directory specification

SoniEx2, 2018-07-26:
https://standards.freedesktop.org/basedir-spec/basedir-spec-latest.html

The XDG Base Directory specification defines base directories for things like config and data files. An example of config files would be enb.conf, epc.conf. An example of data files would be userdb.csv (maybe).

it's nice not having too much clutter in `$HOME`, and being able to easily move most, if not all, config files from one dir to another.

andrepuschmann, 2018-07-27:
Hey, that looks good. Frankly speaking, we were not aware of that spec. Feel free to submit a PR that implements the spec's suggestions.

andrepuschmann, 2019-01-23:
As of 18.12 srsLTE now uses ~/.config/srslte as it's default config folder when running as user application.

closed 2019-01-23 by andrepuschmann