srsran/srsRAN_4G #203: sudo:srsend: command not found

soupnatzi, 2018-08-03:
root@soup-natzi:~/srsLTE# ls
build      CMakeLists.txt            CTestConfig.cmake     lib        srsenb
CHANGELOG  cmake_uninstall.cmake.in  CTestCustom.cmake.in  LICENSE    srsepc
cmake      COPYRIGHT                 debian                README.md  srsue
root@soup-natzi:~/srsLTE# sudo srsenb
sudo: srsenb: command not found

mrlnc, 2018-08-03:
`srsLTE/srsenb` is a folder, not a binary.

Execute `srsLTE/build/srsenb/src/srsenb`.

dylanrchen, 2018-08-05:
Have a similar issue,  and srsenb or srsue are not in the build folder
here is what is inside my build folder: 
CMakeCache.txt 
cmake_install.cmake
CPackConfig.cmake        
CTestCustom.cmake    
DartConfiguration.tcl  lib       
srsepc                 
srslte_install_configs.sh
CMakeFiles      
cmake_uninstall.cmake  
CPackSourceConfig.cmake  
CTestTestfile.cmake  
install_manifest.txt   
Makefile  
SRSLTEbuildinfo.cmake  
Testing

dylanrchen, 2018-08-06:
Just found out this is because i do not have the bladerf driver

andrepuschmann, 2018-08-06:
If no RF driver are found, the UE and eNB applications are not built. That's on purpose.

closed 2018-08-06 by andrepuschmann