srsran/srsRAN_4G #204: Alignment of LIBLTE_BIT_MSG_STRUCT

cpetersgit, 2018-08-06:
LIBLTE_BIT_MSG_STRUCT is defined in liblte_common.h.  As is this worked fine when compiled for x86 on Linux.  However, my messages to the MME were corrupted when compiling for ARM on a custom board.  I tracked the issue to the functions liblte_align_up and liblte_align_up_zero which both assume the input pointer is 64 bit aligned.  If not the bit data is shifted in the messages produced.

My solution was to add 

} LIBLTE_BIT_MSG_STRUCT __attribute__((__aligned__(8)));

This now works on both my Linux and ARM builds the same.

cpetersgit, 2018-08-06:
I had a bunch of the required underscores in my post, but they were removed for some reason.

andrepuschmann, 2018-08-06:
Thanks for this.

closed 2018-10-05 by andrepuschmann