srsran/srsRAN_4G #205: Licensed frequency conflict problem ?

Raw1mage, 2018-08-06:
I was testing srsenb with bx210 operating in band3 overlapped with the commercially licensed band.
Huawei dongle works, but Android phones cannot attach.
I installed "network cell info lite" on Android to observe the signal strength from the eNB, and found issues.

The eNB generates valid strong signal (-60dBm) only for 2 seconds after the initialization, and soon dropped below -140 dBm which is unusable for any phone. 

The odd thing is that Huawei dongle can keep the connection and transmit/receive normally.
However none of the android phones can work with such a weak signal.

My question is, is srseNB dropping the signal strength because it detects the conflict with commercial eNB ?

Raw1mage, 2018-08-08:
This issue is directed less to srsLTE enb.conf
all kinds of variable settings cannot change the phenomenon of weak signals.
Nevertheless, if anyone ever solved the similar situation, a little light is appreciated.

Raw1mage, 2018-08-10:
Anyway, my uhd B210 has trouble delivering strong power to establish a stable connection.   I hope there's at least a gain monitoring tool for tuning.

andrepuschmann, 2018-09-20:
There is no tx power reduction whatsoever in srsENB. Need to reinvestigate whats happening there. Please reopen if needed.

closed 2018-09-20 by andrepuschmann