srsran/srsRAN_4G #206: iphone problem

MrArabi, 2018-08-11:
Hi
i have test LG Nexus 5 and threre was no problem with connection and service's but when i want to connect  iphone 7 to the srslte the iphone event cant discover the network.
i have B210 as RF front and Sysmocom SIM Card.
thank you

andrepuschmann, 2018-08-13:
Hey,
we've never tested with Apple devices. Maybe worth trying to post on the mailing list to see if anybody there can help. Is there any sign of activity in the eNB/EPC logs?

MrArabi, 2018-08-13:
Hi
no the iphone wouldnt even send rach!!!

andrepuschmann, 2018-08-14:
Have you tried with different cell bandwidths?

Raw1mage, 2018-08-21:
My iphone5s works fine with srseNB + nextEPC + USRP B210 antenna.
The only problem is that the antenna generates too weak signal that can barely attachable.
Nevertheless, the throughput is up to 18 MBps (from fast.com)

andrepuschmann, 2018-10-05:
Closing issue due to lack of information. We can't reproduce it either.

closed 2018-10-05 by andrepuschmann

authentic-killer, 2021-09-03:
> Hi
> i have test LG Nexus 5 and threre was no problem with connection and service's but when i want to connect iphone 7 to the srslte the iphone event cant discover the network.
> i have B210 as RF front and Sysmocom SIM Card.
> thank you

what frequeny u had tested nexus 5? i am not able to connect through nexus 5 but it is working with samsung tab