srsran/srsRAN_4G #220: srslte and bladeRF Version: 2018.08-rc2-1ppa1~xenial

Jean-Gaby, 2018-08-21:
Hello there,

There is a problem with the new version of the software bladerf.
I think the type of variables had changed but I do not know how to make the changes.

What I have done:
tutorial

Where does it block?
to this order:

$ make

`[ 51%] Building C object lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_blade_imp.c.o`

```
/home/opale/Bureau/srsLTE/lib/src/phy/rf/rf_blade_imp.c: In function ‘rf_blade_set_rx_freq’:
/home/opale/Bureau/srsLTE/lib/src/phy/rf/rf_blade_imp.c:371:58: error: passing argument 3 of ‘bladerf_get_frequency’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   bladerf_get_frequency(handler->dev, BLADERF_MODULE_RX, &f_int);
                                                          ^
In file included from /home/opale/Bureau/srsLTE/lib/src/phy/rf/rf_blade_imp.c:27:0:
/usr/local/include/libbladeRF.h:1064:15: note: expected ‘uint64_t * {aka long unsigned int *}’ but argument is of type ‘uint32_t * {aka unsigned int *}’
 int CALL_CONV bladerf_get_frequency(struct bladerf *dev,
               ^
/home/opale/Bureau/srsLTE/lib/src/phy/rf/rf_blade_imp.c: In function ‘rf_blade_set_tx_freq’:
/home/opale/Bureau/srsLTE/lib/src/phy/rf/rf_blade_imp.c:389:58: error: passing argument 3 of ‘bladerf_get_frequency’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   bladerf_get_frequency(handler->dev, BLADERF_MODULE_TX, &f_int);
                                                          ^
In file included from /home/opale/Bureau/srsLTE/lib/src/phy/rf/rf_blade_imp.c:27:0:
/usr/local/include/libbladeRF.h:1064:15: note: expected ‘uint64_t * {aka long unsigned int *}’ but argument is of type ‘uint32_t * {aka unsigned int *}’
 int CALL_CONV bladerf_get_frequency(struct bladerf *dev,
               ^
```

What is your bladerf package version ?

Version: 2018.08-rc2-1ppa1~xenial

I think I change the version of bladerf

What do you think ?

Thank you and have a good day

acama, 2018-08-28:
Running into similar issue:

[ 88%] Building CXX object lib/src/phy/rf/CMakeFiles/srslte_rf.dir/uhd_c_api.cpp.o
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c: In function 'rf_blade_set_rx_freq':
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c:371:58: error: passing argument 3 of 'bladerf_get_frequency' from incompatible pointer type [-Werror=incompatible-pointer-types]
   bladerf_get_frequency(handler->dev, BLADERF_MODULE_RX, &f_int);
                                                          ^
In file included from /usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c:27:0:
/usr/include/libbladeRF.h:1117:15: note: expected 'bladerf_frequency * {aka long unsigned int *}' but argument is of type 'uint32_t * {aka unsigned int *}'
 int CALL_CONV bladerf_get_frequency(struct bladerf *dev,
               ^
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c: In function 'rf_blade_set_tx_freq':
/usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c:389:58: error: passing argument 3 of 'bladerf_get_frequency' from incompatible pointer type [-Werror=incompatible-pointer-types]
   bladerf_get_frequency(handler->dev, BLADERF_MODULE_TX, &f_int);
                                                          ^
In file included from /usr/src/srsLTE/lib/src/phy/rf/rf_blade_imp.c:27:0:
/usr/include/libbladeRF.h:1117:15: note: expected 'bladerf_frequency * {aka long unsigned int *}' but argument is of type 'uint32_t * {aka unsigned int *}'
 int CALL_CONV bladerf_get_frequency(struct bladerf *dev,
               ^
cc1: some warnings being treated as errors
lib/src/phy/rf/CMakeFiles/srslte_rf.dir/build.make:134: recipe for target 'lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_blade_imp.c.o' failed
make[2]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_blade_imp.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:2905: recipe for target 'lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all' failed
make[1]: *** [lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

ArshanKhanifar, 2018-09-12:
I had the same issue: 
```
[ 40%] Building C object lib/src/phy/rf/CMakeFiles/srslte_rf.dir/rf_blade_imp.c.o
/home/vagrant/shared/srsLTE/lib/src/phy/rf/rf_blade_imp.c: In function 'rf_blade_set_rx_freq':
/home/vagrant/shared/srsLTE/lib/src/phy/rf/rf_blade_imp.c:371:58: error: passing argument 3 of 'bladerf_get_frequency' from incompatible pointer type [-Werror=incompatible-pointer-types]
   bladerf_get_frequency(handler->dev, BLADERF_MODULE_RX, &f_int);
                                                          ^
In file included from /home/vagrant/shared/srsLTE/lib/src/phy/rf/rf_blade_imp.c:27:0:
/usr/local/include/libbladeRF.h:1160:15: note: expected 'bladerf_frequency * {aka long unsigned int *}' but argument is of type 'uint32_t * {aka unsigned int *}'
 int CALL_CONV bladerf_get_frequency(struct bladerf *dev,
               ^
/home/vagrant/shared/srsLTE/lib/src/phy/rf/rf_blade_imp.c: In function 'rf_blade_set_tx_freq':
/home/vagrant/shared/srsLTE/lib/src/phy/rf/rf_blade_imp.c:389:58: error: passing argument 3 of 'bladerf_get_frequency' from incompatible pointer type [-Werror=incompatible-pointer-types]
   bladerf_get_frequency(handler->dev, BLADERF_MODULE_TX, &f_int);
                                                          ^
In file included from /home/vagrant/shared/srsLTE/lib/src/phy/rf/rf_blade_imp.c:27:0:
/usr/local/include/libbladeRF.h:1160:15: note: expected 'bladerf_frequency * {aka long unsigned int *}' but argument is of type 'uint32_t * {aka unsigned int *}'
 int CALL_CONV bladerf_get_frequency(struct bladerf *dev,
               ^
cc1: some warnings being treated as errors
```
but @jggithub2017 's diff fixed it for me, thanks!

MKassem0, 2018-09-13:
Hi ArshanKhanifar,

I would like to check how do solve this issue? I had the same problem

Thanks

ArshanKhanifar, 2018-09-13:
@MKassem0 just apply @jggithub2017's diff.
If you don't know how, it's simple enough that you can even do it manually. 
```
--- a/lib/src/phy/rf/rf_blade_imp.c
+++ b/lib/src/phy/rf/rf_blade_imp.c
@@ -360,11 +360,11 @@ srslte_rf_info_t *rf_blade_get_info(void *h)
```
this means that on `line 360` of `b/lib/src/phy/rf/rf_blade_imp.c`, you'll have to replace the lines that start with `-` with the lines that start with `+`. 

For a more thorough explanation [check this](https://stackoverflow.com/questions/2529441/how-to-read-the-output-from-git-diff).

MKassem0, 2018-09-13:
@ArshanKhanifar Thank you very much, that perfectly solves the problem