srsran/srsRAN_4G #223: multiple enb's

NateJ3, 2018-08-24:
Has anyone had success connecting multiple eNodeB's simultaneously using SRS?

Thanks,
Nate

closed 2018-08-24 by ismagom

NateJ3, 2018-08-27:
Was there a response to this issue?  I'm hoping someone can help advise how
to address multiple enb's at the same time.

Thanks,
Nate

Nate Johnson
Field Service Manager
Corrections.com
15 Mill Wharf Plaza, Scituate, MA 02066
617-471-4445 (office)
401-439-5484 (cell)
njohnson@corrections.com

ismagom, 2018-08-27:
I closed this issue because it's not an issue. For questions use the list please.

NateJ3, 2018-08-27:
I'm sorry, I'm new to the site.  Can you please direct me to 'the list.'

Thanks,
Nate

Nate Johnson
Field Service Manager
Corrections.com
15 Mill Wharf Plaza, Scituate, MA 02066
617-471-4445 (office)
401-439-5484 (cell)
njohnson@corrections.com

suttonpd, 2018-08-27:
Hi Nate, the mailing list details can be found here: http://www.softwareradiosystems.com/mailman/listinfo/srslte-users

NateJ3, 2018-08-27:
> softwareradiosystems.com/mailman/listinfo/srslte-users

Great, thank you

Nate Johnson
Field Service Manager
Corrections.com
15 Mill Wharf Plaza, Scituate, MA 02066
617-471-4445 (office)
401-439-5484 (cell)
njohnson@corrections.com