srsran/srsRAN_4G #224: Intermittent Transmission on LTE FDD Signals

jamuhn, 2018-09-04:
We are having an issue with intermittent transmission times viewed when looking at our LTE B205mini signal in the time domain. An image is attached with the signal shown in the top right hand corner. Anyone have any ideas why this might be happening? Also, how can we adjust this behavior in order to achieve a constant power transmission?

Thanks!

Jimmy

[image]

ismagom, 2018-09-05:
Hi Jimmy, 

Can you provide the ue.log generated with log.all_level=info setting in ue.conf please?

andrepuschmann, 2018-09-05:
Or is this generated with pdsch_enb? In general, more detailed information about the signal generation, etc. is needed?

jamuhn, 2018-09-05:
I'm using the srsenb. I'll attach our enb.conf. There are no errors or anything noted in the logs. But, I'm not very familiar with what the LTE signal looks like in the time domain. Is this expected behavior? Let me know what else you might need.

[enb.txt](https://github.com/srsLTE/srsLTE/files/2353542/enb.txt)

frankist, 2018-09-05:
Hey Jimmy,

Can you send the logs too? You can find it in "/tmp/enb.log".

Also, just a caution that you will never get fully constant transmit power
in the time domain, due to reference signalling and variable cell load.

Regards,
Francisco

jamuhn, 2018-09-05:
Hi Francisco,

The log is attached. Let me know if you would like a certain log level. There is not much there. Do you think the ratio of transmitting 25% of the time and off 75% of the time we are seeing makes sense?

-Jimmy
[enb.log](https://github.com/srsLTE/srsLTE/files/2354343/enb.log)

andrepuschmann, 2018-09-20:
Spectrum looks ok. Time scale for time domain plot is a 1/th ms so one subframe altogether. Depending on the subframe you're looking at this may be reference symbols and/or sync signals. All good.

closed 2018-09-20 by andrepuschmann