srsran/srsRAN_4G #227: ue cell nof_prb change on enb restart

jgiovatto, 2018-09-06:
Hi Folks,
Happened to notice if I restart enb w/o restarting a previously attched ue the cell nof_prb is not updated. 

Luckily console logs threw me a clue ...
Invalid message length for format 1A
Invalid message length for format 1A
Invalid message length for format 1A
Invalid message length for format 1A
Invalid message length for format 1A
Invalid message length for format 1A

Work around below ...

```
-- a/srsue/src/phy/phch_worker.cc
+++ b/srsue/src/phy/phch_worker.cc
@@ -180,6 +180,11 @@ bool phch_worker::set_cell(srslte_cell_t cell_)
     srslte_ue_ul_set_cfo_enable(&ue_ul, true);
 
     cell_initiated = true;
+  } else {
+    if(cell.nof_prb != cell_.nof_prb) {
+      log_h->console("Cell %d nof_prb %d -> %d\n", cell.id, cell.nof_prb, cell_.nof_prb);
+      cell.nof_prb = cell_.nof_prb;
+    }
   }
   ret = true;
 unlock:
```
Found Cell:  PCI=1, PRB=50, Ports=1, CFO=0.0 KHz, sfn_offset 0
Cell 1 nof_prb 25 -> 50

Ill leave it to the authors to decide if this is a valid use case or not.
Best
Joe

andrepuschmann, 2018-09-07:
Thanks Joe for reporting this. There are a few more considerations to take into account but we'll look into it.

andrepuschmann, 2019-01-21:
Closing this. Will be taken care of in private issue list.

closed 2019-01-21 by andrepuschmann