srsran/srsRAN_4G #241: cmake erro while building srsue

patilmanojh, 2018-09-26:
root@ub18-ts1:~/srsLTE/srsue/build# cmake ../
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at src/phy/CMakeLists.txt:29 (install):
  install TARGETS given no ARCHIVE DESTINATION for static library target
  "srsue_phy".

CMake Error at src/mac/CMakeLists.txt:23 (install):
  install TARGETS given no ARCHIVE DESTINATION for static library target
  "srsue_mac".

CMake Error at src/upper/CMakeLists.txt:33 (install):
  install TARGETS given no ARCHIVE DESTINATION for static library target
  "srsue_upper".

CMake Error at src/CMakeLists.txt:49 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target "srsue".

-- No post-build-UE command defined
CMake Error at src/CMakeLists.txt:61 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target "srsue".

-- No post-build command defined
CMake Error at CMakeLists.txt:52 (install):
  install FILES given no DESTINATION!

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.10)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/root/srsLTE/srsue/build/CMakeFiles/CMakeOutput.log".
root@ub18-ts1:~/srsLTE/srsue/build# 

I am using "Linux ub18-ts1 4.15.0-33-generic #36-Ubuntu SMP "  (Ubuntu 18.04.1 LTS )
and cmake version 3.10.2
I also tried to add ARCHIVE DESTINATION in /src/*/CMakeLists.txt but issue still persist.
install(TARGETS srsue_phy DESTINATION ${LIBRARY_DIR} ARCHIVE DESTINATION ${LIBRARY_DIR})

patilmanojh, 2018-09-27:
Yes I created build directory in top source directory and compiled installed srslte but it only creates srsepc binary. srsue and srsenb are not getting compiled from top i.e. srslte directory

andrepuschmann, 2018-09-27:
Please check the CMake output regarding that. It means that you don't have any RF drivers installed (or at least CMake doesnt pick them up).