srsran/srsRAN_4G #242: USRP N210 Support

grant-h, 2018-09-28:
From the research I have performed online, the N210 has a fixed clock rate of 100MHz (100 Ms/S). This is unlike the B series devices which have a variable clock rate, allowing for easy integer decimation of the samples to LTE preferred rates.

OpenLTE appears to have a working fix for the N210 TX path: https://github.com/osh/openlte/blob/01251fb2ae1ae18c6cced21e0eee9d7c5e2205bf/LTE_fdd_dl_scan/src/LTE_fdd_dl_scan_flowgraph.cc#L246-L251
It involves a FIR filter + rational resampler to properly resample 25 MS/s (100 Ms/s / 4) down to 15.36 Ms/s (10 MHz LTE bandwidth, Nrb = 50)

Given that srsLTE purposefully re-wrote the PHY to avoid GNU radio blocks, how could a similar fix be deployed in this repo? Would something like the below work?

```c++
    srslte_resample_arb_t r;
    srslte_resample_arb_init(&r, rate, 0);
    int n_out = srslte_resample_arb_compute(&r, in, out, N);
```
where rate is some ratio applied to the actual sample rate to bring it into LTE spec.

closed 2019-01-21 by andrepuschmann

grant-h, 2019-01-23:
I'm not a signal processing guy and I'm not really sure how I would go about doing that and validating it. Where and which files would you recommend I change? It's hard to know the right place to change that won't adversely affect timing. Also, how would I know when my change actually worked?

paulUFAM, 2022-12-11:
Hello @grant-h, sorry for the late interaction, but were you able to solve your n210 clock rate problem with the modification proposed above? I'm having the same difficulty even though I already know in advance that srsLTE doesn't support it. I'm waiting for your answer...