srsran/srsRAN_4G #243: NAS tx_count mismatch in lossy environment.

jgiovatto, 2018-10-02:
Hi Folks,
I have (4) ue's each bouncing between (2) enb nodes by applying pathloss on a 1 min interval.
There is also roughly 10% loss on the links at any given time.
I have noticed that the epc will eventually fail the integrity check due to tx_count mismatch, the epc always has a lower "local" count than the "remote". I've applied these changes and it seems to "fix" the issue by not bumping the tx_count until some sort of response it detected.

I am not claiming this is 100% correct it is mainly for informational purposes.

[nas_tx_count_increment.patch.txt](https://github.com/srsLTE/srsLTE/files/2438910/nas_tx_count_increment.patch.txt)

alvasMan, 2018-10-03:
Hi Joe --
Thanks a lot for reporting this, could you send the ue logs so I can have a look? Also the eNB and the EPC pcap?

There was an issue with NAS count mismatch after a service request, but that is already fixed in the next version.

I'm trying to figure out whether this is the same issue or something different.

jgiovatto, 2018-10-03:
Here are the steps:
0) set rf link
1) ue attach enb
2) send ul traffic and stop
3) break rf link
4) set rf link
5) send ul traffc and stop (reconnect success)
6) break rf link
7) send ul traffic, nas retries to connect  (Generating service request tx_count=3  ... then 4
8) set rf link
9) epc Short integrity check failure. Local: count=3, [31 b0 ed 31], Received: count=4, [97 db]
10 ue stuck in WAIT_CON_RECONF
[nas_tx_count.tar.gz](https://github.com/srsLTE/srsLTE/files/2443154/nas_tx_count.tar.gz)

attached logs and patch to print tx_count along the way.

andrepuschmann, 2018-12-18:
We've reworked the counter handling in the new release quite a bit. So I am closing this for now without knowing if it is really fixed. So feel free to re-open if needed. Thanks

closed 2018-12-18 by andrepuschmann