srsran/srsRAN_4G #246: About the sample rate

liychee, 2018-10-08:
Hi, 
When I set the n_prb(Number of Physical Resource Blocks)=100 in enb.conf. I notice that the real sample rate is 20.34M, which means the bandwith is actully 15M? 

I find that in \lib\src\phy\common\phy_common.c the FORCE_STANDARD_RATE is not defined(in the code, if it is defined, it means use_standard_rates=true and n_prb=100 corresponses to the 30.72M sample rate) I modified it, but the srsLTE cannot get random access successful. It always shows Warning: Detected Radio-link Failure
Thank you very much

andrepuschmann, 2018-10-08:
This define gets set by CMake, if it is run with "-DUSE_LTE_RATES=True". If you have issues running srsLTE with this config, please post full logs, etc along with it. But better use the list for this.

closed 2018-10-08 by andrepuschmann

liychee, 2018-10-08:
Thank you very much.
I have tried the method you provide. But it still cannot work. Below is the debug message from the Console window and log file

`linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.000.HEAD-0-g6e1ac3fc

Reading configuration file ue.conf...

Built in Release mode using commit b27c68a on branch master.

---  Software Radio Systems LTE UE  ---

Opening RF device with 1 RX antennas...
Opening USRP with args: type=b200,master_clock_rate=30.72e6
-- Detected Device: B210
-- Operating over USB 3.
-- Detecting internal GPSDO.... Found a generic NMEA GPS device
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 30.720000 MHz... 
-- Actually got clock rate 30.720000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
Waiting PHY to initialize...
...
Attaching UE...
Searching cell in DL EARFCN=3600, f_dl=940.0 MHz, f_ul=895.0 MHz
.
Found Cell:  PCI=1, PRB=100, Ports=1, CFO=-0.1 KHz

Warning TX/RX time offset for sampling rate 30720000 KHz not calibrated. Using interpolated value

Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=25, ra-rnti=0x2
Random Access Transmission: seq=13, ra-rnti=0x2
Random Access Transmission: seq=9, ra-rnti=0x2
Random Access Transmission: seq=6, ra-rnti=0x2
Random Access Transmission: seq=0, ra-rnti=0x2
Random Access Transmission: seq=51, ra-rnti=0x2
Random Access Transmission: seq=33, ra-rnti=0x2
Random Access Transmission: seq=1, ra-rnti=0x2
Random Access Transmission: seq=6, ra-rnti=0x2
Random Access Transmission: seq=20, ra-rnti=0x2
Warning: Detected Radio-Link Failure
Random Access Transmission: seq=14, ra-rnti=0x2
Random Access Transmission: seq=16, ra-rnti=0x2
Random Access Transmission: seq=38, ra-rnti=0x2
Random Access Transmission: seq=0, ra-rnti=0x2
Random Access Transmission: seq=27, ra-rnti=0x2
Random Access Transmission: seq=19, ra-rnti=0x2
Random Access Transmission: seq=2, ra-rnti=0x2
Random Access Transmission: seq=31, ra-rnti=0x2
Random Access Transmission: seq=37, ra-rnti=0x2
Random Access Transmission: seq=7, ra-rnti=0x2
Warning: Detected Radio-Link Failure
Random Access Transmission: seq=7, ra-rnti=0x2
Random Access Transmission: seq=31, ra-rnti=0x2
Random Access Transmission: seq=2, ra-rnti=0x2
Random Access Transmission: seq=36, ra-rnti=0x2
Random Access Transmission: seq=29, ra-rnti=0x2
Random Access Transmission: seq=20, ra-rnti=0x2
Random Access Transmission: seq=10, ra-rnti=0x2
Random Access Transmission: seq=19, ra-rnti=0x2
Random Access Transmission: seq=5, ra-rnti=0x2
Random Access Transmission: seq=44, ra-rnti=0x2
Warning: Detected Radio-Link Failure`

[... 27 lines cut ...]
17:08:44.721668 [MAC ] [E] [07416] RA response not received within the response window
17:08:44.751664 [MAC ] [E] [07446] RA response not received within the response window
17:08:44.752671 [MAC ] [E] [07447] RA:    RxErr:  Maximum number of transmissions reached (10)
17:08:44.752713 [RRC ] [W] Detected Radio-Link Failure
17:08:47.458478 [RRC ] [W] Could not estblish connection. Deallocating dedicatedInfoNAS PDU
17:08:47.458521 [NAS ] [E] Could not establish RRC connection
17:08:47.458750 [NAS ] [E] Could not attach in attach request
17:08:47.481663 [MAC ] [E] [10177] RA response not received within the response window
17:08:47.511664 [MAC ] [E] [10207] RA response not received within the response window
17:08:47.541663 [MAC ] [E] [10237] RA response not received within the response window
17:08:47.571665 [MAC ] [E] [00027] RA response not received within the response window
17:08:47.601664 [MAC ] [E] [00057] RA response not received within the response window
17:08:47.631663 [MAC ] [E] [00087] RA response not received within the response window
17:08:47.661664 [MAC ] [E] [00117] RA response not received within the response window
17:08:47.691664 [MAC ] [E] [00147] RA response not received within the response window
17:08:47.721664 [MAC ] [E] [00177] RA response not received within the response window
17:08:47.751664 [MAC ] [E] [00207] RA response not received within the response window
17:08:47.752663 [MAC ] [E] [00208] RA:    RxErr:  Maximum number of transmissions reached (10)
17:08:47.752688 [RRC ] [W] Detected Radio-Link Failure
17:08:50.460165 [RRC ] [W] Could not estblish connection. Deallocating dedicatedInfoNAS PDU
17:08:50.460206 [NAS ] [E] Could not establish RRC connection
17:08:50.460212 [NAS ] [E] Could not attach in attach request
17:08:50.466240 [RF  ] [W] Underflow
Closing log
`

liychee, 2018-10-08:
Here is message in the enb.log 

`
17:08:44.846331 [PHY0] [E] [07475] Removing user: rnti=0xb8 does not exist
17:08:47.566829 [MAC ] [W] [10196] SCHED: discarting TB 0 pid=4, tti=10196, maximum number of retx exceeded (4)
17:08:47.578373 [RRC ] [W] User rnti=0xb9 expired RRCConnectionSetup deadline: 0:45718>0:40000 us
17:08:47.578448 [PHY0] [E] [10207] Removing user: rnti=0xb9 does not exist
17:08:47.627903 [MAC ] [W] [00017] SCHED: discarting TB 0 pid=1, tti=17, maximum number of retx exceeded (4)
17:08:47.638474 [RRC ] [W] User rnti=0xba expired RRCConnectionSetup deadline: 0:44794>0:40000 us
17:08:47.638544 [PHY1] [E] [00026] Removing user: rnti=0xba does not exist
17:08:47.656812 [MAC ] [W] [00046] SCHED: discarting TB 0 pid=6, tti=46, maximum number of retx exceeded (4)
17:08:47.668555 [RRC ] [W] User rnti=0xbb expired RRCConnectionSetup deadline: 0:45959>0:40000 us
17:08:47.668625 [PHY0] [E] [00057] Removing user: rnti=0xbb does not exist`

andrepuschmann, 2018-10-08:
Try to run the UE with "--time_adv_nsamples=X" and play around with valuen between 30-250 maybe.

liychee, 2018-10-09:
Thank you very much!
I have adjusted the time_adv_nsamples in UE(B210) to a proper value and it works in 20Mhz.
However, when I tried to connect phone(HUAWEI P9) to the EnodeB(B210), I cannot set the time_adv_nsamples in the phone. I tried to set this parameters in EnodeB, but it did not works.
Even in the EnodeB, nothing is showed in the Console window and log file. It seems that the phone can not find the EnodeB