srsran/srsRAN_4G #248: Error when specifying device name when having multiple devices

michalakisk, 2018-10-10:
When setting device name argument in the enb conf after running once successfully the second time i
receive this error

USRP reported the following error: None
LLLLLError receiving from UHD: 30

After trying to start srsenb again i receive the following error on both devices

Error opening RX stream: 30
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args name=USRP1

Error opening RX stream: 30
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args name=USRP2

if i remove argument and use default i get

Error opening RX stream: 30
No compatible RF frontend found
Error opening RF device
Failed to find device auto with args auto

This seems to happen consistently inside containers and seems to be slightly more stable but
problem still occurs when trying it directly on the physical host

On the machine we have 2 USRP's

When the USRP's are unplugged they return back to normal and they continue working, also when srs is giving errors
if i execute uhd_find_device or uhd_usrp_probe both usrps are shown without problem.

--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    type: b200
    name: USRP2
    serial: 3136364
    product: B205mini

--------------------------------------------------
-- UHD Device 1
--------------------------------------------------
Device Address:
    type: b200
    name: USRP1
    serial: 3131080
    product: B205mini

I am using srslte version 18.06_1 and uhd driver 3.14 with latest images

xavierarteaga, 2018-10-10:
Hi @michalakisk,
We have not only reproduced it in our lab but also we found that `benchmark_rate` returns with an error after running it a couple of times.

```/usr/local/lib/uhd/examples/benchmark_rate --tx_rate 3.84e6 --rx_rate 3.84e6 --random```

So, I would say it is a UHD issue.

I recommend you to use UHD version 3.9 LTS. 

BTW, you do not need to unplug the USB cable. You can just reset the USRP running `/usr/local/lib/uhd/utils/b2xx_fx3_utils --reset-device`

closed 2018-10-10 by andrepuschmann

michalakisk, 2018-10-11:
Thank you @xavierarteaga