srsran/srsRAN_4G #250: Sounding reference signal is not support？

liychee, 2018-10-13:
Hi,
I am confused by some setting about Sounding Reference Signal in the EnodeB. In \srsenb\src\phy.cc line 85: `workers_common.pucch_cfg.srs_configured   = false;`

I know that  the SRS is configured in SIB2, but I also cannot find the parameters about SRS such as the srs-Bandwidth or the period of SRS in sib.conf.example？

Does this mean that our srslte does not support Sounding reference signal?

However, I can find the implementation of  SRS in the code, such as SRS generation(function : srslte_refsignal_srs_gen).  So, how can I do to make this SRS works?

Thank you very very much

ismagom, 2018-10-13:
Hi,

SRS is supported in the UE but not in the eNodeB.

liychee, 2018-10-14:
Thank you,
I know that SRS is implemented on the UE side, but it seems that eNodeB is needed to trigger it. In the example, I find that no SRS is triggered in the interaction between eNodeB and UE. How can I configure eNodeB to make UE to send SRS?

liychee, 2018-10-14:
Specifically, according to the 36.211,36.213,36.331, the configuration of SRS is in the SIB2 of eNodeB. (SIB2 mainly accomplishes the following configures: rach, bcch, pcch, prach, pdsch, pusch, pucch, SOUNDING, powercontrol configure. )However, in sib. conf, we can find other configurations, except no SOUNDING configuration. At the same time, we also found in enb_cfg_parser.cc that there is also no corresponding parser for SOUNDING configuration.
Is this a bug?

andrepuschmann, 2018-10-15:
As mentioned above, SRS is currently not supported in the eNB so there is not config option for it.

closed 2018-10-15 by andrepuschmann

YuexinZhao, 2018-10-15:
When will the SRS function in the eNodeB be implemented? Thanks in advance! @ @ismagom @andrepuschmann

ArrrNeo, 2019-05-20:
I am also looking for setting for enabling SRS. Looks like this bug was closed with a commit. However I am unable to find the related config. Can you please tell if this was implemented from eNB side and what is the config to trigger that.

andrepuschmann, 2019-05-21:
It's not supported in the eNB.

ArrrNeo, 2019-05-21:
thanks for the quick reply.