srsran/srsRAN_4G #254: SRB1 integrity check error on multiple ue attach r18.09

jgiovatto, 2018-10-17:
Hi folks,
When 4 ue's rf link is enabled at the same time I see this error usually on 1 ue.
This puts the ue into wait_conn_reconf state.
If I stagger the links by 1 sec all is well.

I was not seeing this in the previous release.

```
11:54:12.959002 [RRC ] [I] Sending Security Mode Complete
11:54:12.959009 [PDCP] [I] TX SRB1 SDU, SN: 3, do_integrity = true, do_encryption = false
             0000: 28 00
11:54:12.959114 [RLC ] [I] SRB1 Tx SDU (7 B, tx_sdu_queue_len=1)
             0000: 03 28 00 7f 08 d9 82
11:54:12.960307 [MAC ] [I] [02971] SR:    Signalling PHY sr_counter=1 
11:54:12.990388 [MAC ] [I] [02985] BSR:   Type Regular, Format Short, Value=2,0,0,0
11:54:12.990781 [RLC ] [I] SRB1 Tx status PDU - ACK_SN = 3, N_nack = 0
11:54:12.990802 [MAC ] [I] [02985] UL 5:  New TX, RV=0, TBS=533, RNTI=73
11:54:12.998757 [MAC ] [I] [02990] SR:    Signalling PHY sr_counter=1
11:54:13.008114 [MAC ] [I] [02994] BSR:   Type Regular, Format Short, Value=10,0,0,0
11:54:13.008147 [RLC ] [I] SRB1 Tx PDU SN=3 (9 B)
             0000: a0 03 03 28 00 7f 08 d9 82
11:54:13.008163 [MAC ] [I] [02994] BSR:   Type Padding, Format Long, Value=0,0,0,0
11:54:13.008171 [MAC ] [I] [02994] UL 6:  New TX, RV=0, TBS=22, RNTI=73
11:54:13.028463 [MAC ] [I] [03004] DL 0 (TB 0):  newTX tbs=105, rv=0, ack=OK, ndi=0 (1), tti=3004 (2969)
11:54:13.028633 [MAC ] [I] [03004] Delivering PDU for lcid=1, 2 bytes
11:54:13.028807 [RLC ] [I] SRB1 Rx control PDU
             0000: 00 10
11:54:13.028823 [RLC ] [I] SRB1 Rx Status PDU: ACK_SN = 4, N_nack = 0
11:54:13.028830 [MAC ] [I] [03004] Delivering PDU for lcid=1, 99 bytes
11:54:13.028836 [RLC ] [I] SRB1 Rx data PDU SN=3 (97 B)
             0000: 03 20 06 02 51 38 65 5b c2 38 08 3a 10 09 f0 30
             0010: 00 07 88 80 00 38 00 ea 90 0e 08 08 38 38 33 9b
11:54:13.028856 [RLC ] [I] SRB1 Status packet requested through polling bit
11:54:13.028862 [RLC ] [I] SRB1 Rx SDU (97 B)
             0000: 03 20 06 02 51 38 65 5b c2 38 08 3a 10 09 f0 30
             0010: 00 07 88 80 00 38 00 ea 90 0e 08 08 38 38 33 9b
11:54:13.028872 [PDCP] [I] RX SRB1 PDU, do_integrity = true, do_encryption = true
             0000: 03 20 06 02 51 38 65 5b c2 38 08 3a 10 09 f0 30
             0010: 00 07 88 80 00 38 00 ea 90 0e 08 08 38 38 33 9b
11:54:13.028882 [PDCP] [I] RX SRB1 PDU (decrypted)
             0000: 03 20 06 02 51 38 65 5b c2 38 08 3a 10 09 f0 30
             0010: 00 07 88 80 00 38 00 ea 90 0e 08 08 38 38 33 9b
11:54:13.029032 [PDCP] [E] MAC mismatch (expected)
             0000: 67 2f fa 46
11:54:13.029037 [PDCP] [E] MAC mismatch (found)
             0000: 29 bd 20 35
11:54:13.029043 [PDCP] [E] SRB1 Dropping PDU
             0000: 03 20 06 02 51 38 65 5b c2 38 08 3a 10 09 f0 30
             0010: 00 07 88 80 00 38 00 ea 90 0e 08 08 38 38 33 9b
11:54:13.030427 [MAC ] [I] [03005] BSR:   Type Regular, Format Short, Value=2,0,0,0
11:54:13.030454 [RLC ] [I] SRB1 Tx status PDU - ACK_SN = 4, N_nack = 0
```

jgiovatto, 2018-10-23:
First verified all inputs to `security_128_eia()` were identical on `integrity_generate()` and `integirty_verify()` which should yield the same MAC at the ENB/UE respectively as it usually does.

Added logs to the begin and end of `integrity_generate()` and it looks like calls may be "stacking up" so
calls to `integrity_generate()`  may not be re-entrant.

Most obvious attempt to use MAC variable from caller in function `snow3g_f9()` vs `static u8 MAC_I` w/o success. This was not totally surprising since the MAC values actual and expected we unique to the receiving UE and NOT showing up in other the UE logs as was the case in P/R #234, so the theory is the inputs are being "stepped on" vs the result ?

Short term solution added mutex to `integrity_generate()` and `integrity_verify()` has shown to be  successful. Not sure if the is the correct place for this or should higher level call `write_sdu` be protected ?

Patch attached below:

[pdcp_integrity_mutex.patch.txt](https://github.com/srsLTE/srsLTE/files/2506450/pdcp_integrity_mutex.patch.txt)

andrepuschmann, 2018-10-24:
Thanks for reporting this Joe. We'll look into it.

andrepuschmann, 2018-12-17:
Should be addressed in next release.

closed 2018-12-17 by andrepuschmann