srsran/srsRAN_4G #256: SIB13 not applied if it is not already present in rrc::configure_serving_cell

JonathanRennison, 2018-10-23:
In `rrc::configure_serving_cell`, SIB2 and SIB13 are applied if they are present using `rrc::apply_sib2_configs` and `rrc::apply_sib13_configs` respectively.

In the case where they are not present, they are fetched and when received are processed by `rrc::handle_sib2` and `rrc::handle_sib13` respectively.
`rrc::handle_sib2` calls `rrc::apply_sib2_configs` as expected, however `rrc::handle_sib13` does nothing.

The effect of this is that PMCH decoding and MBMS does not work if the SIB13 had not already been received when `rrc::configure_serving_cell` is called. Calling `rrc::apply_sib13_configs` from `rrc::handle_sib13` fixed the issue here.

andrepuschmann, 2018-10-24:
Thanks for reporting this Jonathan. We'll look into it.

andrepuschmann, 2018-12-18:
Fixed in next release.

closed 2018-12-18 by andrepuschmann