srsran/srsRAN_4G #259: Incorrect mbsfn_per_frame calculated in srsue::phch_common::is_mch_subframe

JonathanRennison, 2018-10-23:
With reference to the line:
`uint32_t mbsfn_per_frame = mcch->pmch_infolist_r9[0].pmch_config_r9.sf_alloc_end_r9/liblte_rrc_mch_scheduling_period_r9_num[mcch->pmch_infolist_r9[0].pmch_config_r9.mch_schedulingperiod_r9];`

`sf_alloc_end_r` is not required to be the same as the number of allocated MCH subframes x scheduling period.
In testing the transmitted value was found to be MCH subframes x scheduling period - 1. Presumably the -1 is for the MCCH subframe.
As a result the `if(sf_alloc_idx <= mch_period_stop) {` test is wrong for frames where not all MCH subframes are used.

andrepuschmann, 2018-10-24:
Thanks for reporting this Jonathan. We'll look into it.

andrepuschmann, 2019-04-25:
Fixed in new release.

closed 2019-04-25 by andrepuschmann