srsran/srsRAN_4G #260: srsue::dl_harq_entity::mutex left locked causing deadlock if subframe fails CRC check

JonathanRennison, 2018-10-23:
In `srsue::dl_harq_entity::new_grant_dl` the mutex `srsue::dl_harq_entity::mutex` is locked.
It is unlocked in `srsue::dl_harq_entity::tb_decoded`.
If the subframe fails its CRC check, for example due to poor reception, `srsue::dl_harq_entity::tb_decoded` is never called and the UE deadlocks on the next attempt to acquire the mutex.

andrepuschmann, 2018-10-24:
Thanks for reporting this Jonathan. We'll look into it.

JonathanRennison, 2018-10-24:
To add some information on this, in our testing this problem has occurred when the warning
> DL PID %d: Size of grant changed during a retransmission

is emitted in `srsue::dl_harq_entity::new_grant_dl`.
Though not all cases where this warning was emitted resulted in a deadlock.

andrepuschmann, 2018-12-14:
Not an issue, mutex is unlocked later on, see tb_decoded_()

closed 2018-12-14 by andrepuschmann