srsran/srsRAN_4G #262: Compilation of latest release on ARM/Odroid system

HughAndersonNUS, 2018-11-02:
Hi,
I just tried to compile the latest srsLTE on both an Ubuntu 16.04 and and Ubuntu 18.04.1 Odroid/ARM system, but got an error related to the SSE. When I do the make, I get:
```
[ 10%] Building C object lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o
/home/odroid/srsLTE/lib/src/phy/fec/turbodecoder.c: In function ‘srslte_tdec_init_manual’:
/home/odroid/srsLTE/lib/src/phy/fec/turbodecoder.c:170:22: error: ‘sse16_win_impl’ undeclared (first use in this function); did you mean ‘sse_impl’?
       h->dec16[0] = &sse16_win_impl;
                      ^~~~~~~~~~~~~~
                      sse_impl
/home/odroid/srsLTE/lib/src/phy/fec/turbodecoder.c:170:22: note: each undeclared identifier is reported only once for each function it appears in
/home/odroid/srsLTE/lib/src/phy/fec/turbodecoder.c:178:21: error: ‘sse8_win_impl’ undeclared (first use in this function); did you mean ‘sse16_win_impl’?
       h->dec8[0] = &sse8_win_impl;
                     ^~~~~~~~~~~~~
                     sse16_win_impl
lib/src/phy/fec/CMakeFiles/srslte_fec.dir/build.make:302: recipe for target lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o' failed
```

The cmake looked like this:

```
odroid@odroid:~/srsLTE/build$ cmake ../
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-3.10.105-138
-- CMAKE_SYSTEM_PROCESSOR: armv7l
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find mkl (missing: MKL_LIBRARIES MKL_CORE MKL_ILP MKL_SEQ) 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.7
-- FFTW3F LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found fftw3f: /usr/lib/arm-linux-gnueabihf/libfftw3f.so  
-- FFT_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find POLARSSL (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/arm-linux-gnueabihf/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/arm-linux-gnueabihf/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MBEDTLS: /usr/lib/arm-linux-gnueabihf/libmbedcrypto.so  
-- Checking for module 'libpcsclite'
--   No package 'libpcsclite' found
-- Could NOT find PCSCLITE (missing: PCSCLITE_LIBRARY PCSCLITE_INCLUDE_DIR) 
-- PCSC LIBRARIES: PCSCLITE_LIBRARY-NOTFOUND
-- PCSC INCLUDE DIRS: PCSCLITE_INCLUDE_DIR-NOTFOUND
-- UHD LIBRARIES /usr/lib/arm-linux-gnueabihf/libuhd.so
-- UHD INCLUDE DIRS /usr/include
-- Found UHD: /usr/lib/arm-linux-gnueabihf/libuhd.so  
-- Checking for module 'libbladeRF'
--   Found libbladeRF, version 1.7.2-2016.06-1-ppaxenial
-- Looking for bladerf_get_board_name in bladeRF
-- Looking for bladerf_get_board_name in bladeRF - not found
-- libbladeRF found but not compatible. Upgrade your driver or use SoapySDR.
-- FINDING SOAPY.
-- Checking for module 'SoapySDR'
--   Found SoapySDR, version 0.6.1
-- Found libSOAPYSDR: /usr/include/SoapySDR, /usr/lib/arm-linux-gnueabihf/libSoapySDR.so
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   program_options
-- SRSGUI LIBRARIES /usr/local/lib/libsrsgui.so
-- SRSGUI INCLUDE DIRS /usr/local/include
-- Found SRSGUI: /usr/local/lib/libsrsgui.so  
-- Performing Test HAVE_SSE
-- Performing Test HAVE_SSE - Failed
-- Performing Test HAVE_AVX
-- Performing Test HAVE_AVX - Failed
-- Performing Test HAVE_AVX2
-- Performing Test HAVE_AVX2 - Failed
-- Performing Test HAVE_FMA
-- Performing Test HAVE_FMA - Failed
-- Performing Test HAVE_AVX512
-- Performing Test HAVE_AVX512 - Failed
-- Performing Test HAVE_ERROR_INCOMPATIBLE
-- Performing Test HAVE_ERROR_INCOMPATIBLE - Failed
-- have ARM
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Failed
-- CMAKE_C_FLAGS is  -march=native -Wall -Wno-comment -Wno-write-strings -Winline -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype-limits -std=c99 -D_GNU_SOURCE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -mfpu=neon -march=native -DIS_ARM -DHAVE_NEON
-- CMAKE_CXX_FLAGS is  -march=native -Wall -Wno-comment -Wno-reorder -Wno-unused-but-set-variable -Wno-unused-variable -Wformat -Wmissing-field-initializers -Wtype-limits -std=c++03 -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE
-- clang-tidy not found.
-- Using install prefix: /usr/local
-- Building for version: 18.9.0
--    examples will be installed.
-- No post-build command defined
-- Building with srsUE
-- No post-build-UE command defined
-- No post-build command defined
-- Building with srsENB
-- Found LibConfig++: /usr/lib/arm-linux-gnueabihf/libconfig++.so
-- static LibConfig++ path: /usr/lib/arm-linux-gnueabihf/libconfig++.a
-- Found LibConfig: /usr/lib/arm-linux-gnueabihf/libconfig.so
-- static LibConfig path: /usr/lib/arm-linux-gnueabihf/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/arm-linux-gnueabihf/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- Found SCTP: /usr/lib/arm-linux-gnueabihf/libsctp.so  
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/arm-linux-gnueabihf/libconfig++.so
-- static LibConfig++ path: /usr/lib/arm-linux-gnueabihf/libconfig++.a
-- Found LibConfig: /usr/lib/arm-linux-gnueabihf/libconfig.so
-- static LibConfig path: /usr/lib/arm-linux-gnueabihf/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/arm-linux-gnueabihf/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/odroid/srsLTE/build
odroid@odroid:~/srsLTE/build$ 
```

Can anyone shed light on this?

Cheers Hugh

andrepuschmann, 2018-11-02:
Hugh, we are aware of this issue and are working on a fix for master. For the time being, please use the previous release (18.06) which works well on ARM. Thanks

HughAndersonNUS, 2018-11-03:
OK - I tried 18.06, and 18.06.1 releases, and it got past that error, and ended up with the following errors:
```
[ 18%] Building C object lib/src/phy/mimo/CMakeFiles/srslte_mimo.dir/layermap.c.o
[ 18%] Building C object lib/src/phy/mimo/CMakeFiles/srslte_mimo.dir/precoding.c.o
In file included from /home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/mat.h:31:0,
                 from /home/odroid/srsLTE-release_18_06_1/lib/src/phy/mimo/precoding.c:36:
/home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/simd.h: In function ‘srslte_simd_f_abs’:
/home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/simd.h:514:21: error: incompatible type for argument 1 of ‘vqabsq_s32’
   return vqabsq_s32(a);
                     ^
In file included from /home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/simd.h:39:0,
                 from /home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/mat.h:31,
                 from /home/odroid/srsLTE-release_18_06_1/lib/src/phy/mimo/precoding.c:36:
/usr/lib/gcc/arm-linux-gnueabihf/7/include/arm_neon.h:5692:1: note: expected ‘int32x4_t’ but argument is of type ‘simd_f_t’
 vqabsq_s32 (int32x4_t __a)
 ^~~~~~~~~~
In file included from /home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/mat.h:31:0,
                 from /home/odroid/srsLTE-release_18_06_1/lib/src/phy/mimo/precoding.c:36:
/home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/simd.h: In function ‘srslte_simd_cf_neg’:
/home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/simd.h:994:6: error: ‘simd_cf_t {aka struct float32x4x2_t}’ has no member named ‘re’
   ret.re = srslte_simd_f_neg(a.re);
      ^
/home/odroid/srsLTE-release_18_06_1/lib/include/srslte/phy/utils/simd.h:994:31: error: ‘simd_cf_t {aka struct float32x4x2_t}’ has no member named ‘re’
   ret.re = srslte_simd_f_neg(a.re);
                               ^
... and so on.
```
This is on a Ubuntu 18.04.1 Odroid/ARM system, with all the prerequisites I think. The cmake looked like this:
```
odroid@odroid:~/srsLTE-release_18_06_1/build$ cmake ../
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-3.10.105-138
-- CMAKE_SYSTEM_PROCESSOR: armv7l
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find mkl (missing: MKL_LIBRARIES MKL_CORE MKL_ILP MKL_SEQ) 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.7
-- FFTW3F LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found fftw3f: /usr/lib/arm-linux-gnueabihf/libfftw3f.so  
-- FFT_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
[... 56 lines cut ...]
-- Found LibConfig: /usr/lib/arm-linux-gnueabihf/libconfig.so
-- static LibConfig path: /usr/lib/arm-linux-gnueabihf/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/arm-linux-gnueabihf/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- Found SCTP: /usr/lib/arm-linux-gnueabihf/libsctp.so  
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/arm-linux-gnueabihf/libconfig++.so
-- static LibConfig++ path: /usr/lib/arm-linux-gnueabihf/libconfig++.a
-- Found LibConfig: /usr/lib/arm-linux-gnueabihf/libconfig.so
-- static LibConfig path: /usr/lib/arm-linux-gnueabihf/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/arm-linux-gnueabihf/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/odroid/srsLTE-release_18_06_1/build
odroid@odroid:~/srsLTE-release_18_06_1/build$ make
```

Cheers Hugh

andrepuschmann, 2018-11-18:
Hey Hugh,
we'll fix the ARM issue in the next release. Sorry for the inconvenience. And yes, LimeSDR is only supported through SoapySDR.
Thanks
Andre

closed 2018-12-14 by andrepuschmann