srsran/srsRAN_4G #264: RLC issue

IlshatMkII, 2018-11-16:
Hello SRS Team!

I want to report about some RLC layer issues in eNB.

[-] As I understand, RLC timers' methods can be simultaneously called from 2 different threads. I don't think that it's good.

[-] In RLC AM there are several problems with segment PDU:
1. You need to send segments NACK in status PDU whenever it's possible. Let's illustrate it with example. 
UE sends large PDU and radio conditions are poor. eNB received almost the whole PDU except small     segment. In your program you send NACK for whole PDU instead of NACK for small segment. And UE starts to retransmit the whole PDU. Probability of PDU loss increases dramatically.

2. RLC can receive segments in random order. Let's illustrate it with example. 
UE sends segment#1 at TTI=k, then sends segment#2 at TTI=k+1. ENB fails do decode PUSCH at TTI=k, but successfully decodes PUSCH at TTI=k+1. That means segment#2 will be received by RLC earlier. In your program you always try to add new segment to the end of list. So your program will discard segment#1 when it comes.

3. UE can send "partially overlapping" segments. Let's illustrate it with example.
eNB receives segment[0 byte - 100 byte] and segment [200 byte - 300 byte]. Then eNB sends Status Report and UE transmits segment [95 byte - 210 byte]. So eNB should extract only [100 byte - 200 byte] from new segment and perform concatenation. 

These RLC AM issues are very critical especially for DRBs.

suttonpd, 2018-11-16:
Hi,
Thanks for flagging these issues! We'll work to address them for our upcoming releases. 
If you're available to discuss further, it would be great to make contact. I'm paul@softwareradiosystems.com.

Best regards,
Paul

andrepuschmann, 2018-11-20:
Hey, we've looked at the issues you've raised and here is our initial analysis. Regarding issue:
  1. You're right, we don't support segment NACKs and instead NACK the whole PDU. We may add support for that in one of the upcoming releases. 
  2. The AM receiver does handle retransmitted PDU and also PDU segments, see for example resegment_test_1() in rlc_am_test.cc (change the skipped SN to 0 and you'll test both cases)
  3. Confirmed. We indeed do not handle overlapping segments. This will definitely be fixed in the next release.

Regarding the timer issue, I am not sure if I understand your question correctly. The RLC timer handler is only called from the MAC timer thread and access to shared variables is protected. Where do you see the race-condition?

IlshatMkII, 2018-11-21:
Hello, SRS Team!

**The 2nd issue**
_rlc_am::rlc_am_rx::add_segment_and_check()_ is called every time when PDU segment is received. It performs incorrect actions. Some more examples:

1. UE sends PDU segment#1 [0 byte - 50 byte] at TTI = k, then segment#2 [50 byte - 100 byte] at TTI = k + 1. Let's suppose that eNB fails to decode PUSCH at TTI = k, but succeeds at TTI = k + 1 (segment#2) and at TTI = k + 8 (HARQ Retx of segment#1). That means segment#2 reaches RLC AM first, but it will be discarded because header.so != 0. 

2. UE sends segment#1 [0 byte - 30 byte], segment#2 [30 byte - 50 byte] and segment#3 [50 byte - 70 byte]. Let's suppose that eNB receives segments in the following order: 1 -> 3 -> 2. Segment#3 will be discarded because its header.so != 31. 

**Some suggestions for RLC AM PDU segments**
Maybe it is better to use the following structure for PDU segments: pointer to one srslte::byte_buffer_t and list of segment headers. Every time when new segment is received, RLC AM will push header into list and copy segment bytes into byte_buffer_t. I think using of byte_buffer_t for each segment is wasteful.

**RLC Timers**
In function _rlc_am::rlc_am_rx::handle_data_pdu()_  T_Reordering could be stopped/restarted according to specification (by pdu_process_thread). But at the same time MAC timer thread could trigger T_Reordering expiry before timer is stopped/restarted.  Yes, it will be blocked until  _rlc_am::rlc_am_rx::handle_data_pdu()_ finishes (because of mutex). But then it will perform expiry related actions, even if T_Reordering was stopped / restarted. 
I'm not sure whether it is critical. 

**Some suggestions for Timers**
Maybe it is better to add thread to RLC layer, which will periodically check and perform expiry related actions.

andrepuschmann, 2018-11-23:
Hey, looked at the examples again you gave and I can confirm that we're actually not handling retransmitted segments correctly. The issue isn't critical in the sense that it doesn't cause data corruption but the performance could be degraded, as you've already pointed out. Same goes for timers, you're right but I don't see it to be critical. But also this will be fixed in one of the upcoming releases. I've moved both issues into our internal bug-tracking and will therefore close the issue here. Thanks again for pointing that out. Much appreciated

closed 2018-11-23 by andrepuschmann