srsran/srsRAN_4G #265: srsLTE 18.09 fails to build when SIMD is unavailable (18.06 worked just fine)

rubund, 2018-11-18:
When building without any of the SSE, AVX, and "-march=native" options, the latest release of srsLTE fails to build. This worked fine with release 18.06.

Below is the console output where it fails:

`
[ 15%] Building C object lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o
cd /home/ruben/devel/maintain_stretch/deb_srslte/srslte/obj-x86_64-linux-gnu/lib/src/phy/fec && /usr/bin/cc -DHAVE_MBEDTLS -DHAVE_PCSC -I/usr/include/PCSC -I/home/ruben/devel/maintain_stretch/deb_srslte/srslte/obj-x86_64-linux-gnu/lib/include -I/home/ruben/devel/maintain_stretch/deb_srslte/srslte/lib/include  -g -O2 -fdebug-prefix-map=/home/ruben/devel/maintain_stretch/deb_srslte/srslte=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -Wall -Wno-comment -Wno-write-strings -Winline -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype-limits -std=c99 -D_GNU_SOURCE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -fPIC   -Werror=incompatible-pointer-types -fvisibility=hidden -o CMakeFiles/srslte_fec.dir/turbodecoder.c.o   -c /home/ruben/devel/maintain_stretch/deb_srslte/srslte/lib/src/phy/fec/turbodecoder.c
/home/ruben/devel/maintain_stretch/deb_srslte/srslte/lib/src/phy/fec/turbodecoder.c: In function ‘srslte_tdec_init_manual’:
/home/ruben/devel/maintain_stretch/deb_srslte/srslte/lib/src/phy/fec/turbodecoder.c:170:22: error: ‘sse16_win_impl’ undeclared (first use in this function); did you mean ‘sse_impl’?
       h->dec16[0] = &sse16_win_impl;
                      ^~~~~~~~~~~~~~
                      sse_impl
/home/ruben/devel/maintain_stretch/deb_srslte/srslte/lib/src/phy/fec/turbodecoder.c:170:22: note: each undeclared identifier is reported only once for each function it appears in
/home/ruben/devel/maintain_stretch/deb_srslte/srslte/lib/src/phy/fec/turbodecoder.c:178:21: error: ‘sse8_win_impl’ undeclared (first use in this function); did you mean ‘sse_impl’?
       h->dec8[0] = &sse8_win_impl;
                     ^~~~~~~~~~~~~
                     sse_impl
make[3]: *** [lib/src/phy/fec/CMakeFiles/srslte_fec.dir/build.make:196: lib/src/phy/fec/CMakeFiles/srslte_fec.dir/turbodecoder.c.o] Error 1
make[3]: Leaving directory '/home/ruben/devel/maintain_stretch/deb_srslte/srslte/obj-x86_64-linux-gnu'
make[2]: *** [CMakeFiles/Makefile2:1678: lib/src/phy/fec/CMakeFiles/srslte_fec.dir/all] Error 2
make[2]: Leaving directory '/home/ruben/devel/maintain_stretch/deb_srslte/srslte/obj-x86_64-linux-gnu'
make[1]: *** [Makefile:166: all] Error 2
make[1]: Leaving directory '/home/ruben/devel/maintain_stretch/deb_srslte/srslte/obj-x86_64-linux-gnu'
dh_auto_build: cd obj-x86_64-linux-gnu && make V=1 -j1 "INSTALL=install --strip-program=true" returned exit code 2
`

Best regards
Ruben

rubund, 2018-11-18:
looks like a duplicate of https://github.com/srsLTE/srsLTE/issues/262 more or less.

andrepuschmann, 2018-11-18:
Is this also on ARM? If so, please use 18.03, we'll fix that for the next release.

rubund, 2018-11-18:
No, it is for Debian official package build. (https://buildd.debian.org/status/package.php?p=srslte) Many of the architectures do not have SIMD (mips, s390x, powerpc, riscv64 etc.)

Btw. Does srsLTE do runtime detection of the availabilitiy of AVX2 and SSE4 etc? Just thinking that also for the amd64 build of srsLTE, we cannot build for anything which is not supported by ALL CPUs of that architecture. Currently, SIMD is disabled completely on all architectures until I have the full overview here. (Using something like "volk" would probably make packaging binaries easier, since it already handles runtime detection)  :)

andrepuschmann, 2018-11-19:
If you have dedicated builds for all architectures, it should be detected ok. If not we may have an issue detected the SIMD extensions. We've never tested on mips, powerpc, etc.
Unfortunately, we don't have runtime detection of SIMD extensions in our SIMD library, but that is something worth thinking about. Maybe a srslte-avx and srslte-avx2 would make sense until this is around.

rubund, 2018-11-24:
> Maybe a srslte-avx and srslte-avx2 would make sense until this is around.

I have implemented this now for the Debian package. Thanks for the idea. A simple shell script checks what extensions are available before selecting one of the executables: srsenb-avx2, srsenb-avx, srsenb-sse4, srsenb-generic (and same for srsue).

> If not we may have an issue detecting the SIMD extensions. We've never tested on mips, powerpc, etc.

For version 18.10, srsenb-generic does not build at all - not on amd64 either. But 18.06 works fine.

andrepuschmann, 2018-11-24:
Thanks for the heads up. This looks good. I see how we can adopt something similar for Ubuntu and maybe include in mainline. In general, I observed that there are quite a few patches with bug-fixes and typos, etc. that you apply only to the Debian build. May I ask why those have not been pushed upstream?

rubund, 2018-11-24:
> May I ask why those have not been pushed upstream?

I think probably only the spelling-fix patch is also relevant for upstream. The others are more specific to the need for the package build. The answer to your question is that I simply haven't gotten there yet.. But now that you already have seen the patches, you are of course free to cherry-pick whatever you want. :)

> I see how we can adopt something similar for Ubuntu 

You are probably also aware of that the package in Debian also goes into the official Ubuntu archive (https://launchpad.net/ubuntu/+source/srslte)  and all other Debian derivatives.

Feel free to pick from the packaging files. It is all AGPL-3+..

andrepuschmann, 2018-12-14:
18.06 was the last release with a generic Turbo-decoder. All newer releases need at least SSE4.2 for x86 or NEON on ARM for the Turbo-decoder. As a consequence, only the EPC is really working on those systems.

closed 2018-12-14 by andrepuschmann